pyocd/pyOCD #95: Writing to floating point registers with GDB does not work properly

c1728p9, 2015-01-26:
Writes to s0-s31 do not work as expected.  Writes to even numbered 's' registers  modify the wrong register.  The value from modifying s[n] is s[n/2].  For example setting the register s6 will modify the value of s3.  Writes to registers d0-d15 do not work as well.

adamgreen, 2015-01-26:
Thanks for catching this.  Fix is in the queue.

0xc0170, 2015-01-26:
Was merged, can you @c1728p9 confirm it's fixed now, and close this if so ?

c1728p9, 2015-01-27:
Yep, it is fixed.  Thanks for the quick fix.  

Out of curiosity, the 'd' registers are doubles and are the combination of two 's' registers right?  Does the feature 'org.gnu.gdb.arm.vfp' in the xml let gdb know that there are two 's'  for every 'd' registers?  I don't see any 's' registers in the xml, but I could have overlooked something.

closed 2015-01-27 by c1728p9

adamgreen, 2015-01-27:
@c1728p9 You are correct, there are two s (single precision) registers for each d (double precision) register.  The 'org.gnu.gdb.arm.vfp' tells GDB that there is a VFP.  The XML must then define the d* registers and then it will automatically set up the s* pseudo registers to properly map over the d* ones.  You can see this in the arm_gdbarch_init(), arm_pseudo_read(), and arm_pseudo_write() routines of GDB's arm-tdep.c source file.

While I suspect that you can map the s* registers over the d* ones in the XML, this isn't done in the target XML examples found in the gdb/features/ sample folder of the GDB source tree and isn't required because of the pseudo register support found in arm-tdep.c.  I pulled the XML for Cortex-M4F targets from that sample folder.