pyocd/pyOCD #106: Most popular pyOCD interface/IDE & readme.md instructions

c1728p9, 2015-03-01:
I was wondering what most people have been using as an interface to pyOCD?  Standalone GDB, Eclipse CDT + GDB, something else?  If you use eclipse, are there any plugins that you find useful?  I was thinking it would be good to add one or more recommended debugging configurations to readme.md.

I use pyOCD along with eclipse and GDB.  I use it through GNU ARM Eclipse's OpenOCD plugin.  This plugin can be configured to start a program when debugging begins (for me this is a .bat file which runs gdb_server.py with parameters).  It also has a reset button, which regular CDT doesn't (though the reset command can be issued through the gdb console of regular eclipse).  I consider this configuration a little unorthodox, but it works well.  I also use the Embedded Systems Register View eclipse plugin to view peripheral registers.

Anyone have any configurations they would recommend?  Think it would be good to add my (or another) configuration to readme.md, so people who are unfamiliar with pyOCD can get debugging without having to go searching everywhere on how to set things up?

0xc0170, 2015-03-02:
This is an excellent question :raised_hands: , I would like to have this documented somewhere (=available for users). For example, I was not aware of systems register view plugin, will need to do my search.

As I had some problems setting proper eclipse, and was time consuming to do it all over again, I started using command line debugging. Can you share you setup? I would start from there and see what people say once we got one configuration.

We can create a doc folder with some docs (might be once transferred to own github page).

c1728p9, 2015-09-08:
Closing this issue since there are basic setup instructions in the readme.  If anyone has ideas for other useful configurations please reopen and comment.

closed 2015-09-08 by c1728p9