pyocd/pyOCD #112: pyOCD fails on 2nd dev.open call on device with non-matching target id

mikew67, 2015-03-20:
**Synopsis:** Running basic_test.py on a new installation of pyOCD on a Mac, with an nRF51-DK connected, yields a "IOError: open failed" error.

**Scenario:**
**MacBook Pro, Core 2 Duo**
Mac OS X "Yosemite" 10.3.3 (pre-release seed 3)
Python 2.7.6
**Nordic nRF51-DK platform**
mbed IF: nrf51822aa_atsam3u2c_if_mbed_V0218_19022015.bin
Enumerates as: 
    Product ID:	0x0204
    Vendor ID:	0x0d28
    Version:	1.00
    Manufacturer:	MBED
**Backtrace on failure**
```
  File "basic_test.py", line 46, in <module>
    with MbedBoard.chooseBoard() as board:
  File "/Users/mwirth/Desktop/MacGitRepos/pyOCD/pyOCD/board/mbed_board.py", line 169, in chooseBoard
    all_mbeds = MbedBoard.getAllConnectedBoards(transport, False, blocking, target_override, frequency)
  File "/Users/mwirth/Desktop/MacGitRepos/pyOCD/pyOCD/board/mbed_board.py", line 122, in getAllConnectedBoards
    all_mbeds = INTERFACE[usb_backend].getAllConnectedInterface(mbed_vid, mbed_pid)
  File "/Users/mwirth/Desktop/MacGitRepos/pyOCD/pyOCD/interface/hidapi_backend.py", line 78, in getAllConnectedInterface
    dev.open(vid, pid)
  File "hid.pyx", line 48, in hid.device.open (hid.c:1263)
IOError: open failed
```

**Diagnostics:**
Code snippet around dev.open call:
```
            try:
                dev.open(vid, pid)
            except AttributeError:
            pass
```

c1728p9, 2015-03-21:
Hi @mikew67 I have this branch to address your first point.  With this you should be able to get the nrf51-dk running: https://github.com/c1728p9/pyOCD/tree/nrf51-dk

0xc0170, 2015-03-21:
Thanks for reporting @mikew67 . As @c1728p9 pointed out, the other nrf chips were not added (their id is missing).

As I can see, there will be added soon :+1:

c1728p9, 2015-05-26:
Hi @mikew67 , the pull request #110 fixed the problem where the same board is opened twice if it is unsupported.  The pull request #142 adds support for the nrf51-dk.  Can this issue be closed, or do you see any further problems?

mikew67, 2015-05-27:
Yes, I'll close this.  Thanks for the followup.

Mike

closed 2015-05-27 by mikew67