pyocd/pyOCD #113: Test binary for each development board rather than each processor

c1728p9, 2015-03-22:
Currently when pyOCD finishes basic_test.py it will program a new image to flash.  Right now this image is determined by the processor type.  The problem I'm concerned with is that several development kits can have the same processor, but have pins routed to different hardware.  Because of this, an image that caused an led to blink on one device could sink current into a button an another.  For this reason would to be good to modify pyOCD to use a test binary for each development board rather than each processor?

0xc0170, 2015-04-16:
board id to target defines targets (which is the board), kl25z is frdm-kl25z.
```
BOARD_ID_TO_TARGET = {
                "0200": "kl25z",
```

And the binary is chosen based on the target (each target has own specific number).

What do you propose? To update targets name to reflect the real names, kl25z -> frdm-kl25z ?

c1728p9, 2015-04-16:
Right now several development boards used the same binary:
```
"1070" -> "nrf51822" -> l1_nrf51822.bin
"9009" -> "nrf51822" -> l1_nrf51822.bin
"9012" -> "nrf51822" -> l1_nrf51822.bin
```

It might make more sense to be structured like this:
```
"1070" -> "Nordic_nRF51-Dongle.bin"
"9009" -> "Nordic_nRF51-DK.bin"
"9012" -> "Seeed_Tiny_BLE.bin"
```

That way each board will have dedicated firmware for its hardware.  Even with this change the target and flash names would stay the same.  There would just be a dedicated binary for each board.

c1728p9, 2015-05-06:
Done in #129

closed 2015-05-06 by c1728p9