pyocd/pyOCD #120: Can't load a binary on OS X 10.10.3

autopulated, 2015-04-16:
Trying to use gdb 'load', with a pyocdgdbserver (the target is a K64F board):

```
> load
Loading section .isr_vector, size 0x198 lma 0x0
Loading section .flash_protect, size 0x10 lma 0x400
Loading section .text, size 0x2b8e8 lma 0x420
Loading section .ARM.extab, size 0xdc lma 0x2bd08
Loading section .ARM.exidx, size 0x1b0 lma 0x2bde4
Loading section .data, size 0xa60 lma 0x2bf94
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 153, in run
    [resp, ack, detach] = self.handleMsg(data)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 220, in handleMsg
    return self.flashOp(msg[2:]), 1, 0
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 380, in flashOp
    self.flash.init()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/flash/flash.py", line 45, in init
    self.target.writeBlockMemoryAligned32(self.flash_algo['load_address'], self.flash_algo['instructions'])
  File "/usr/local/lib/python2.7/site-packages/pyOCD/target/cortex_m.py", line 646, in writeBlockMemoryAligned32
    self.transport.writeBlock32(addr, data[:n/4])
  File "/usr/local/lib/python2.7/site-packages/pyOCD/transport/cmsis_dap.py", line 191, in writeBlock32
    dapTransferBlock(self.interface, len(data), WRITE | AP_ACC | AP_REG['DRW'], data)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/transport/cmsis_dap_core.py", line 227, in dapTransferBlock
    max_pending_reads = interface.getPacketCount()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/interface/interface.py", line 48, in getPacketCount
    return self.packet_count
AttributeError: 'HidApiUSB' object has no attribute 'packet_count'
```

0xc0170, 2015-04-16:
@c1728p9 This was introduced in the commit d2b5c06d4cde08c2ed8a11

c1728p9, 2015-04-16:
Pull request #121 should fix this

0xc0170, 2015-04-16:
Thanks @c1728p9 for looking at this so quickly :+1:

closed 2015-04-20 by 0xc0170