pyocd/pyOCD #125: pyOCD crashes when GDB issues the load command.

bremoran, 2015-04-22:
When I issue ```load``` from the gdb console, pyOCD emits the following block and crashes.
```
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/local/Cellar/python/2.7.8_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 152, in run
    [resp, ack, detach] = self.handleMsg(data)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 219, in handleMsg
    return self.flashOp(msg[2:]), 1, 0
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 360, in flashOp
    self.flashBuilder.addData(write_addr, self.unescape(data[idx_begin:len(data) - 3]))
  File "/usr/local/lib/python2.7/site-packages/pyOCD/flash/flash_builder.py", line 115, in addData
    operation.addr, operation.addr + len(operation.data)))
ValueError: Error adding data - Data at 0x0..0x198 overlaps with 0x0..0x198
```

c1728p9, 2015-04-22:
Thanks for the report.  I just put up #126 to fix this.

0xc0170, 2015-04-22:
@bremoran please retest, the fix was merged

bremoran, 2015-04-22:
I can confirm that this is fixed.

closed 2015-04-23 by 0xc0170