pyocd/pyOCD #127: LPC4088 Quick Start Board

johosemi, 2015-04-23:
Below are very minimally tested changes needed for the EA LPC4088 QSB. Only change from lpc1768 is RAM size, so be carefull!

diff --git a/pyOCD/board/mbed_board.py b/pyOCD/board/mbed_board.py
index 0860a28..a83f3ce 100644
--- a/pyOCD/board/mbed_board.py
+++ b/pyOCD/board/mbed_board.py
@@ -35,6 +35,7 @@ BOARD_ID_TO_TARGET = {
                 "9004": "lpc1768",
                 "1040": "lpc11u24",
                 "1050": "lpc800",
+                "1060": "lpc4088",
                 "1070": "nrf51822",
                 "9009": "nrf51822",
                 "9012": "nrf51822",
diff --git a/pyOCD/flash/__init__.py b/pyOCD/flash/__init__.py
index 4a8c32c..717e19b 100644
--- a/pyOCD/flash/__init__.py
+++ b/pyOCD/flash/__init__.py
@@ -43,6 +43,7 @@ FLASH = {
          'lpc800': Flash_lpc800,
          'lpc11u24': Flash_lpc11u24,
          'lpc1768':  Flash_lpc1768,
+         'lpc4088':  Flash_cortex_m,
          'lpc4330':  Flash_lpc4330,
          'nrf51822': Flash_nrf51822,
          'stm32f103rc': Flash_stm32f103rc,
diff --git a/pyOCD/target/__init__.py b/pyOCD/target/__init__.py
index 37cf61f..1d0a174 100644
--- a/pyOCD/target/__init__.py
+++ b/pyOCD/target/__init__.py
@@ -28,6 +28,7 @@ import target_k20d50m
 import target_lpc800
 import target_lpc11u24
 import target_lpc1768
+import target_lpc4088
 import target_lpc4330
 import target_nrf51822
 import target_stm32f103rc
@@ -49,6 +50,7 @@ TARGET = {
           'lpc800': target_lpc800.LPC800,
           'lpc11u24': target_lpc11u24.LPC11U24,
           'lpc1768': target_lpc1768.LPC1768,
+          'lpc4088': target_lpc4088.LPC4088,
           'lpc4330': target_lpc4330.LPC4330,
           'nrf51822': target_nrf51822.NRF51822,
           'stm32f103rc': target_stm32f103rc.STM32F103RC,
diff --git a/pyOCD/target/target_lpc4088.py b/pyOCD/target/target_lpc4088.py
new file mode 100644
index 0000000..be24bd6
--- /dev/null
+++ b/pyOCD/target/target_lpc4088.py
@@ -0,0 +1,48 @@
+"""
+ mbed CMSIS-DAP debugger
+ Copyright (c) 2006-2015 ARM Limited
+
+ Licensed under the Apache License, Version 2.0 (the "License");
+ you may not use this file except in compliance with the License.
+ You may obtain a copy of the License at
+
+     http://www.apache.org/licenses/LICENSE-2.0
+
+ Unless required by applicable law or agreed to in writing, software
+ distributed under the License is distributed on an "AS IS" BASIS,
+ WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
+ See the License for the specific language governing permissions and
+ limitations under the License.
+"""
+
+from cortex_m import CortexM
+
+class LPC4088(CortexM):
+
+    memoryMapXML =  """<?xml version="1.0"?>
+<!DOCTYPE memory-map PUBLIC "+//IDN gnu.org//DTD GDB Memory Map V1.0//EN" "http://sourceware.org/gdb/gdb-memory-map.dtd">
+<memory-map>
+    <memory type="flash" start="0x0" length="0x10000"> <property name="blocksize">0x1000</property></memory>
+    <memory type="flash" start="0x10000" length="0x70000"> <property name="blocksize">0x8000</property></memory>
+    <memory type="ram" start="0x10000000" length="0x10000"> </memory>
+    <memory type="ram" start="0x20000000" length="0x8000"> </memory>
+</memory-map>
+"""
+
+    def __init__(self, transport):
+        super(LPC4088, self).__init__(transport)
+        self.auto_increment_page_size = 0x1000
+
+    def reset(self, software_reset = False):
+        CortexM.reset(self, False)
+
+    def resetStopOnReset(self, software_reset = False, map_to_user = True):
+        CortexM.resetStopOnReset(self)
+
+        # Remap to use flash and set SP and SP accordingly
+        if map_to_user:
+            self.writeMemory(0x400FC040, 1)
+            sp = self.readMemory(0x0)
+            pc = self.readMemory(0x4)
+            self.writeCoreRegisterRaw('sp', sp)
+            self.writeCoreRegisterRaw('pc', pc)

0xc0170, 2015-04-23:
Could you please send a pull request ?

johosemi, 2015-04-23:
Git is not my friend, so figuring out how to do it will take some time, sorry.

johosemi, 2015-04-23:
Hope I did it right ...

c1728p9, 2017-02-11:
Hi @johosemi this has been done in #252

closed 2017-02-11 by c1728p9