pyocd/pyOCD #134: "IndexError: list index out of range" while loading executable from GDB to FRDM-K64F

autopulated, 2015-05-12:
I can't reproduce this, but thought it might be useful to report anyway:
```
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 153, in run
    [resp, ack, detach] = self.handleMsg(data)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 220, in handleMsg
    return self.flashOp(msg[2:]), 1, 0
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 369, in flashOp
    self.flashData += self.unescape(data[idx_begin:len(data) - 3])
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 437, in unescape
    data[data_idx] = data[data_idx] ^ 0x20
IndexError: list index out of range
```

c1728p9, 2015-05-13:
Thanks for the report.  Judging by the location of the crash, it looks like a gdb flashwrite packet got corrupted, causing the data to end with the escape character, 0x7d.  How often do you see this?  Also is your socket connection from GDB client to host on the same PC, or is it connected via the internet?

autopulated, 2015-05-13:
I've only seen this once. I'm connecting to localhost.

autopulated, 2015-05-20:
This same error just happened to me again (still on localhost). Restarting pyOCD and re-loading the same program worked fine.

c1728p9, 2017-02-11:
Closing due to lack of activity. Please re-open if you see this again.

closed 2017-02-11 by c1728p9