pyocd/pyOCD #137: Error shutting down PyOCD after disconnecting device

autopulated, 2015-05-20:
Not sure if you care about fixing this, but I thought I'd report it just in case:

```
Interrupted while waiting for the program.
Give up (and stop debugging it)? (y or n) y
> Quit
> quit
info: GDB server thread killed
error: exception during uninit
Process Process-1:
Traceback (most recent call last):
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/process.py", line 258, in _bootstrap
    self.run()
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/multiprocessing/process.py", line 114, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/local/lib/python2.7/site-packages/valinor-0.0.6-py2.7.egg/valinor/gdb.py", line 47, in _launchPyOCDGDBServer
    gdb.stop()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 75, in stop
    self.board.uninit()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/board/board.py", line 82, in uninit
    self.transport.uninit()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/transport/cmsis_dap.py", line 129, in uninit
    dapDisconnect(self.interface)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/transport/cmsis_dap_core.py", line 134, in dapDisconnect
    if resp[0] != COMMAND_ID['DAP_DISCONNECT']:
IndexError: list index out of range
```

0xc0170, 2015-05-20:
Of course we do care, thanks for reporting

autopulated, 2015-05-20:
It isn't like you can do anything to recover after I've unplugged the device :stuck_out_tongue_winking_eye:

c1728p9, 2017-02-11:
This issue should be fixed on master. Instead of failing due to list index, an exception is explicitly raised.

closed 2017-02-11 by c1728p9