pyocd/pyOCD #138: ValueError: The device does not support the specified langid

SeppoTakalo, 2015-05-22:
I cloned the pyusb and pyOCD, and installed those.

Connected K64F board.

~~~
sudo python test/gdb_server.py 
Welcome to the PyOCD GDB Server Beta Version 
INFO:root:Waiting for a USB device connected
uncaught exception: The device does not support the specified langid
Traceback (most recent call last):
  File "test/gdb_server.py", line 62, in <module>
    board_selected = MbedBoard.chooseBoard(board_id = option.board_id, target_override = option.target_override, frequency = option.debug_clock_frequency)
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/board/mbed_board.py", line 201, in chooseBoard
    all_mbeds = MbedBoard.getAllConnectedBoards(transport, False, blocking, target_override, frequency)
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/board/mbed_board.py", line 148, in getAllConnectedBoards
    all_mbeds = INTERFACE[usb_backend].getAllConnectedInterface(mbed_vid, mbed_pid)
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/interface/pyusb_backend.py", line 111, in getAllConnectedInterface
    product_name = usb.util.get_string(board, 256, 2)
  File "/usr/local/lib/python2.7/dist-packages/usb/util.py", line 328, in get_string
    raise ValueError("The device does not support the specified langid")
ValueError: The device does not support the specified langid
~~~

0xc0170, 2015-05-22:
I just wonder what firmware does you board have, the latest ?

I tested and it worked, with 
```
Version: 0221
Build:   Jan  7 2015 02:30:03

SeppoTakalo, 2015-05-22:
Check this
https://github.com/walac/pyusb/commit/dac78933f6a6eaf5ae82f48e2f4e7d1733dc2f98

You have extra parameter in
https://github.com/mbedmicro/pyOCD/blob/master/pyOCD/interface/pyusb_backend.py#L111

0xc0170, 2015-05-22:
Seppo thanks for some more context. For reference, this is the PR to pyusb which introduces the error message reported https://github.com/walac/pyusb/pull/99

0xc0170, 2015-05-22:
Thanks for reporting ! I sent a PR which should fix this issue.

closed 2015-05-22 by 0xc0170