pyocd/pyOCD #143: Error loading .ELF via gdb_server on FRDM-K64F

thom-nic, 2015-05-26:
I built some examples from the Freescale KSDK 1.2 and I'm trying to load one in particular (httpsrv) but it crashes the GDB server when I do.  Here's the output from gdbserver:

```
± python test/gdb_server.py
Welcome to the PyOCD GDB Server Beta Version 
INFO:root:new board id detected: 02400221C3163E593EC8C3E1
INFO:root:board allows 5 concurrent packets
INFO:root:DAP SWD MODE initialised
INFO:root:IDCODE: 0x2BA01477
INFO:root:K64F not in secure state
INFO:root:6 hardware breakpoints, 4 literal comparators
INFO:root:CPU core is Cortex-M4
INFO:root:FPU present
INFO:root:4 hardware watchpoints
INFO:root:GDB server started at port:3333
INFO:root:One client connected!
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/Users/thom/dev/sandbox/kinetis/pyOCD/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 152, in run
    [resp, ack, detach] = self.handleMsg(data)
  File "/Users/thom/dev/sandbox/kinetis/pyOCD/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 219, in handleMsg
    return self.flashOp(msg[2:]), 1, 0
  File "/Users/thom/dev/sandbox/kinetis/pyOCD/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 360, in flashOp
    self.flashBuilder.addData(write_addr, self.unescape(data[idx_begin:len(data) - 3]))
  File "/Users/thom/dev/sandbox/kinetis/pyOCD/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 412, in unescape
    data[data_idx] = data[data_idx] ^ 0x20
IndexError: list index out of range
```

And on the gdb client side:
```
$ $ARMGCC_DIR/bin/arm-none-eabi-gdb httpsrv.elf         
GNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20140731-cvs

(...... GDB header output ......)

(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x00003124 in CLOCK_HAL_GetPll0Clk (base=0x4006a000)
    at /Users/thom/dev/sandbox/kinetis/ksdk/KSDK_1.2.0/platform/hal/src/mcg/fsl_mcg_hal.c:525
525	    if (!(MCG_BRD_S_PLLST(base) || MCG_BRD_C5_PLLCLKEN0(base)))
(gdb) load
Loading section .interrupts, size 0x400 lma 0x0
Loading section .flash_config, size 0x10 lma 0x400
Loading section .text, size 0x414c0 lma 0x410
Remote connection closed
(gdb)
```

Here's a link to the .elf file in question: http://s000.tinyupload.com/?file_id=28283428874542781329
I've had luck with other sample apps from the KSDK so not sure why this one in particular is causing an issue.  It is a bit large.

0xc0170, 2015-05-26:
Thanks for sharing, this was reported also in #134, but now we have an app to reproduce it with!

thom-nic, 2015-05-26:
Awesome, yeah that binary triggers it pretty consistently.  Sorry for the duplicate issue otherwise :)

c1728p9, 2015-05-27:
Thanks for the report.  I tried to reproduce this with the following GDB versions, but had no luck:
GNU gdb (GNU Tools for ARM Embedded Processors) 7.4.1.20121207-cvs
GNU gdb (GNU Tools for ARM Embedded Processors) 7.8.1.20141128-cvs

Which year and quarter of GNU Tools ARM Embedded are you using?  I'll try downloading the same version and testing to see if I can reproduce the problem.

0xc0170, 2015-05-27:
GNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20140731-cvs

Tested on the latest master (v0.4, with one slight change in setup.py , looks like utility package was not included there, will send PR soon). Loaded 5x , no problems

```
(gdb) load
Loading section .interrupts, size 0x400 lma 0x0
Loading section .flash_config, size 0x10 lma 0x400
Loading section .text, size 0x414c0 lma 0x410
Loading section .ARM, size 0x8 lma 0x418d0
Loading section .init_array, size 0x4 lma 0x418d8
Loading section .data, size 0xa3c lma 0x418dc
Start address 0x410, load size 271128
Transfer rate: 17 KB/sec, 1922 bytes/write.
(gdb)
```

thom-nic, 2015-05-27:
```
$ $ARMGCC_DIR/bin/arm-none-eabi-gdb --version
GNU gdb (GNU Tools for ARM Embedded Processors) 7.6.0.20140731-cvs

$ $ARMGCC_DIR/bin/arm-none-eabi-gcc --version
arm-none-eabi-gcc (GNU Tools for ARM Embedded Processors) 4.8.4 20140725 (release) [ARM/embedded-4_8-branch revision 213147]
```

On Mac OSX 10.10 if that makes any diff...  Thanks!

0xc0170, 2015-05-27:
Same gcc version as mine, which brings me to a question if this is not OS related? Previous report referenced above here + this one was reported on Mac OS. Btw, I have tested on Windows.

c1728p9, 2015-05-27:
I tested on windows as well.  Could you try re-running while logging to a file - "set remotelogfile pyocd.log".  Two or three runs would also be helpful, so I can see if it fails in the same place each time.  Also, which interface are you using?  pyusb or hidapi?

c1728p9, 2015-09-08:
Hi @thom-nic , have you seen this issue recently?  PR #167 might have fixed this.  If so this issue should be closed.

thom-nic, 2015-09-08:
When I filed this I was evaluating the K64 for a project and have since moved on to other stuff so I'm not using the K64 or pyOCD day-to-day.  I'll see if I can get a chance to try it again sometime soon.

c1728p9, 2015-09-08:
Also could you re-post your .elf file that you were having trouble with?  It looks like tinyupload deleted it.

c1728p9, 2017-02-11:
Closing due to lack of activity. Please re-open if you see this again.

closed 2017-02-11 by c1728p9