pyocd/pyOCD #144: GDBServer error "No module named utility.conversion" in 0.4.0

autopulated, 2015-05-27:
```
  File "/usr/local/lib/python2.7/site-packages/valinor-0.0.6-py2.7.egg/valinor/gdb.py", line 26, in _launchPyOCDGDBServer
    from pyOCD.gdbserver import GDBServer
  File "/usr/local/lib/python2.7/site-packages/pyOCD/__init__.py", line 18, in <module>
    import board
  File "/usr/local/lib/python2.7/site-packages/pyOCD/board/__init__.py", line 18, in <module>
    import mbed_board
  File "/usr/local/lib/python2.7/site-packages/pyOCD/board/mbed_board.py", line 22, in <module>
    from board import Board
  File "/usr/local/lib/python2.7/site-packages/pyOCD/board/board.py", line 18, in <module>
    from pyOCD.target import TARGET
  File "/usr/local/lib/python2.7/site-packages/pyOCD/target/__init__.py", line 18, in <module>
    import cortex_m
  File "/usr/local/lib/python2.7/site-packages/pyOCD/target/cortex_m.py", line 23, in <module>
    from ..gdbserver import signals
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/__init__.py", line 18, in <module>
    from gdbserver import GDBServer
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdbserver.py", line 21, in <module>
    from ..utility.conversion import hexStringToIntList, hexEncode, hexDecode
ImportError: No module named utility.conversion
```

```
pip list | grep pyOCD
pyOCD (0.4)
```
(Attempting to debug a STM32F401RE)

autopulated, 2015-05-27:
The `utility` module seems to be missing from the installation: (this remained the case after pip uninstall and reinstall):

```
 ls /usr/local/lib/python2.7/site-packages/pyOCD/
__init__.py  __init__.pyc board        flash        gdbserver    interface    target       transport
```

0xc0170, 2015-05-27:
Ah, just noticed and noted few minutes ago (https://github.com/mbedmicro/pyOCD/issues/143#issuecomment-105891668), will release a new fixed version. 

Thanks

0xc0170, 2015-05-27:
#145 merged with a fix, v0.4.1 pushed.

closed 2015-05-27 by 0xc0170