pyocd/pyOCD #151: Board ID

ohagendorf, 2015-06-04:
Where can I get board IDs?

I can see the list in pyOCD and know that they are compiled into the CMSIS DAP firmware. But I didn't find the information how to get a new one.

Regards
Olaf

0xc0170, 2015-06-05:
There was this question in cmsis-dap or here, I can't locate it at the moment.

The list of ids is in cmsis-dap, https://github.com/mbedmicro/CMSIS-DAP/blob/master/interface/Common/src/board.c

The comment there:
```
#if !defined(BOARD_SECRET) || !defined(BOARD_ID)
// TODO: how to check a string value using the preprocessor...
#error contact support@mbed.org to obtain a secret and board ID
#endif
```

Are you building  a new target interface?

ohagendorf, 2015-06-05:
Thanks a lot for the information. I'll ask there.

No, we are not building a new target. But the CMSIS DAP implementation of X893 seems to run on DISCO_F407 and DISCO_F429 very well. The existing firmware can be used directly with Keil or CoIDE. But for pyOCD one USB HID vendor command to detect the board is missing beside the other USB VID/PID

An STLink v2 does not support a CDC device as CMSIS DAP does. We resoldered several times the STLink mcu to have this functionality also with DISCO_F407 and F429. But replacing STLink firmware with CMSIS DAP is much easier.
The second reason is pyOCD. With this package we can use the same programming interface under Windows and Linux without the need of a specific device driver.

Another idea would be to extend pyOCD with a command line parameter in a way that a target can be choosen manually.

c1728p9, 2015-06-06:
The ability to choose a target manually already exists.  When running gdb_server or flash_tool you can use the '-t <target>' option.

c1728p9, 2015-09-08:
Hi @ohagendorf , have you been able to get a board id, or find another solution to your problem?  If so this issue should be closed.

closed 2015-09-10 by ohagendorf