pyocd/pyOCD #161: Failed "load" command on gdb_test.py

kaizen8501, 2015-06-11:
Hi
I added new target for W7500 of WIZnet MCU. And then I success automated_test.py on pyOCD.

## Get fail message when I enter load command on GNU gdb
After automated_test , I tried to test "gdb_test.py". connection is success to gdb server of pyOCD. But I got a fail message when issue "load" command. This picture is log message when I tested.
Left terminal in this picture is log message when I run gdb_test.py and right terminal is log message of GNU gdb.

[image]

I want to know Why can't success after load command. If analyzer_supported is false in flash_w7500.py, Can't it use gdb for debugging W7500?

And How to solve this problem?

c1728p9, 2015-06-11:
Hi @kaizen8501 , I would recommend using the gdb_server.py in pyOCD\tools.  The file gdb_test.py hasn't been updated in a while and might be out of date.  Also to add extra debug info you could use the "-d debug" argument in gdb_server.py.  

If that still doesn't work could you run the command "set remotelogfile pyocd.log" and then post the output of pyocd.log here?  That way I can see if there are any problems with the gdb communication.

Finally, does this problem only occur when analyzer_supported it set to true?

kaizen8501, 2015-06-11:
Thank you @c1728p9 , As soon as I changed begin_data to 0x20002000 and analyzer_supported to "true", I solved this problem. So I can use load command of GNU gdb.

## Other Question

And then I tried to test using Eclipse IDE but I can't success as below picture.

[image]

Do you know why did it happen? For more information, I attached my debug informtion on Eclipse.
[image]

c1728p9, 2015-06-11:
Glad I could help :smile:. The warning "Invalid mon command 'regs'" is expected since pyOCD does not support this command.  Eclipse doesn't seem to use this information and has better ways to retrieve regsiter data anyway.

The warning "Invalid mon command 'reset init'" is expected as well, but the command will probably be supported in the future.  As a simple workaround you could change the initial reset type to 'halt'.  This can be done in the 'Startup' tab of the 'WebSocket_WizFi250 OpenOCD' debug configuration.

As for the error at address 0xffffffff, this appreas to be from the program crashing, or a blank part.  You'll want to check that code was loaded correctly staring at address 0.  See if you can set a breakpoint early on in the boot sequence.

kaizen8501, 2015-06-16:
Hi @c1728p9. I solved this problem as this document.
[http://www.life4iot.com/2015/06/15/how-to-use-gdb-on-eclipse-with-pyocd/?lang=en](http://www.life4iot.com/2015/06/15/how-to-use-gdb-on-eclipse-with-pyocd/?lang=en) <-- This is URL of my blog.

As a result, I can connect to gdb server of pyOCD from Eclipse IDE successfully.

Additionally, I have one question. I heard that mbed web compiler will support debugger function next plan. I think pyOCD will be included on mbed web compiler. Is it right?

Thank you.

c1728p9, 2015-06-16:
Good to hear you got things working.  Cool blog, I like that you are sharing information about pyOCD :smiley:.  I'm not sure on plans for web debugger support.  @0xc0170 ,do you have any details on this that can be shared?

0xc0170, 2015-06-29:
I'll close this as it's resolved.

@c1728p9 I am not familiar with the plans for this future.

closed 2015-06-29 by 0xc0170