pyocd/pyOCD #163: Missing dependency on cython breaks OS X installation

autopulated, 2015-06-18:
Tested with homebrew python 2.7.9:
```
e106982-mac:~ jamcro01$ pip install pyOCD
Collecting pyOCD
  Using cached pyOCD-0.4.3.zip
Collecting hidapi (from pyOCD)
  Using cached hidapi-0.7.99-5.tar.gz
    Complete output from command python setup.py egg_info:
    Traceback (most recent call last):
      File "<string>", line 20, in <module>
      File "/private/var/folders/yg/wqy54z_s18ng_3q7qndkbg9nszkzrm/T/pip-build-pz8Kjf/hidapi/setup.py", line 4, in <module>
        from Cython.Distutils import build_ext
    ImportError: No module named Cython.Distutils
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/yg/wqy54z_s18ng_3q7qndkbg9nszkzrm/T/pip-build-pz8Kjf/hidapi
```

Try installing cython:
```
e106982-mac:~ jamcro01$ pip install cython
Collecting cython
  Downloading Cython-0.22-cp27-none-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl (3.7MB)
    100% |████████████████████████████████| 3.7MB 120kB/s 
Installing collected packages: cython
Successfully installed cython-0.22
```

Then everything works:
```
e106982-mac:~ jamcro01$ pip install pyOCD
Collecting pyOCD
  Using cached pyOCD-0.4.3.zip
Collecting hidapi (from pyOCD)
  Using cached hidapi-0.7.99-5.tar.gz
Collecting intelhex (from pyOCD)
  Downloading intelhex-2.0-py2.py3-none-any.whl (42kB)
    100% |████████████████████████████████| 45kB 1.5MB/s 
Installing collected packages: hidapi, intelhex, pyOCD
  Running setup.py install for hidapi
  Running setup.py install for pyOCD
Successfully installed hidapi-0.7.99.post5 intelhex-2.0 pyOCD-0.4.3
```

posborne, 2015-06-18:
IIRC, there are some tricky dependencies in order to build Cython from source.  I see a couple ways to handle this:

1. Try to get `cython-hidapi` to include generated .c files with their sdist (I have done this on projects before and it effectively removes the dependency on Cython for people who just want to use your package)
2. Document how to install Cython's build dependencies and add it to the instructions
3. Tell users to install cython (however they choose; e.g. brew or pip or whatever) prior to installing pyOCD.

posborne, 2015-07-02:
It appears that cython includes a bdist_wheel for mac x86_64 which should cover most users.  Given that, I think it would be reasonable to add

autopulated, 2015-07-03:
+1 for that solution.

mpg, 2015-07-10:
Just ran into this issue, ran `pip install cython` before trying again, everything works like a charm now. (brew doesn't seem to have a formula for cython: brew search shows only a couple closed pull requests for cython).

I don't know a thing about python packaging, but from my layman point of view, adding a dependency on cython on OS X looks like a good short-term solution. Getting the `cython-hidapi` maintainers to include generated C files as suggested by Paul above looks even better, but I guess that's something on which we have less control.

0xc0170, 2015-09-01:
Should be resolved in the latest release (v0.5.1)

janjongboom, 2016-01-05:
@0xc0170  I ran into this today when installing yotta...

c1728p9, 2016-01-08:
@janjongboom Can you post the error message you get when you install yotta

janjongboom, 2016-01-11:
My logs are here: https://github.com/janjongboom/yotta-playground/blob/master/install-notes.md

```
    building 'hid' extension
    error: unknown file type '.pxd' (from 'chid.pxd')

    ----------------------------------------
Command "/usr/bin/python -c "import setuptools, tokenize;__file__='/private/var/folders/23/tsgskn7n75d8jc07yf2fs2n08gb5z6/T/pip-build-67POr5/hidapi/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/23/tsgskn7n75d8jc07yf2fs2n08gb5z6/T/pip-kSpd_1-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/23/tsgskn7n75d8jc07yf2fs2n08gb5z6/T/pip-build-67POr5/hidapi
```

c1728p9, 2016-03-25:
@janjongboom does pyOCD still do this when you try to install it?

janjongboom, 2016-03-26:
@c1728p9 Wouldn't know. This was on a clean OSX install. My OSX is not clean anymore :p

immunda, 2016-11-30:
I'm still finding the same error.

c1728p9, 2017-02-11:
@immunda can you post the error your getting and the steps to reproduce?

andrewleech, 2017-05-07:
For reference, I just had this issue on osx. It appears to be caused by older python version. 

It happened when installing into a virtualenv created with python 2.7.10
I'm pretty sure it's the system python.
For reference, cython installed from source, not from wheel.

I rebuilt the virtualenv with python 2.7.13 installed with homebrew and it's worked fine now. cython came from wheel this time.