pyocd/pyOCD #164: typo in setup.py? linux2 instead of linux

arunpersaud, 2015-06-18:
Hi

just wondering if 

    install_requires = dict(
        win32=['pyWinUSB'],
        linux2=['pyUSB'],
        darwin=['hidapi'] 
    )

the 2 for linux is a typo? I get a key error from

    install_requires[sys.platform]

geky, 2015-06-18:
Hi @arunpersaud, can you find what sys.platform is on your platform? On my machine it's 'linux2' but it looks like it's based on where python was compiled:
https://docs.python.org/3/library/sys.html#sys.platform

arunpersaud, 2015-06-18:
Hi

I'm trying to build the package on the openSUSE build server. The full log is for example available here:

https://build.opensuse.org/package/live_build_log/home:apersaud:branches:devel:languages:python3/python3-pyOCD/openSUSE_13.2/x86_64

on 

https://build.opensuse.org/package/show/home:apersaud:branches:devel:languages:python3/python3-pyOCD

you can also get the build results for some of the distro-versions and architectures.

Just tried it on my own desktop (also openSUSE) and there I get 'linux2' for pyhton 2.7 and 'linux' for python3. So it might be an issue with the version of python (all the builds are also for python3).

Arun

geky, 2015-06-18:
Well this should at least take care of the platform name mismatch: #165

closed 2015-06-29 by 0xc0170