pyocd/pyOCD #177: pyusb dependency failing

autopulated, 2015-08-03:
Hi,

I've had some reports that the pyusb dependency is failing to install on Linux (Ubuntu 14.04.2 LTS, pip 1.5.6, python 2.7.6)

```
 Could not find a version that satisfies the requirement pyusb (from pyOCD>=0.3,<1.0->valinor>=0.0.0,<1.0->yotta) (from versions: 1.0.0a2, 1.0.0a2, 1.0.0a3, 1.0.0a3, 1.0.0b1, 1.0.0b2)
  Some externally hosted files were ignored (use --allow-external to allow).
```

It isn't obvious to me why this is. It's possible that pip isn't liking the non-semantic versions that pyusb has. pip does seem to be extremely old on this system...

c1728p9, 2015-08-13:
Hi @autopulated , I had this same problem.  Even with an updated version of pip, the command "pip install pyusb" fails on Ubuntu.  I was able to get pyusb to install by specifying the version explicitly " pip install pyusb==1.0.0b2".

0xc0170, 2015-09-01:
This should be resolved in the v0.5.x.

closed 2015-09-01 by 0xc0170