pyocd/pyOCD #183: Release needs to be pushed to fix hidapi for Mac

bridadan, 2015-08-31:
Currently its impossible to install yotta with `pip install yotta` due to its dependency on pyOCD and pyOCD's dependency on Cython. This commit should fix this issue: https://github.com/mbedmicro/pyOCD/commit/712ad33d780bc41a02ee67a8b49d4829b4b74ac8.

A release needs to be created and pushed to PyPi.

@0xc0170 Any chance this can be taken care of? Or can anyone else push this to PyPi?

0xc0170, 2015-08-31:
It will be released soon

0xc0170, 2015-09-01:
Done, https://github.com/mbedmicro/pyOCD/releases/tag/v0.5.1

bridadan, 2015-09-01:
Thanks for pushing this. I just tested this on a clean Mac and it unfortunately still fails when installing through pip (even though it works fine if you install from cloning the git repo). I don't really have any other ideas on how to solve this other than installing Cython beforehand manually. Yotta would have to do the same. Any other ideas?

c1728p9, 2015-09-01:
It might work if added to both install_requires and setup_requires.  Sounds like setuptools won't actually install packages in setup_requires unless they are also in install_requires:
http://pythonhosted.org/setuptools/setuptools.html#developer-s-guide
```
(Note: projects listed in setup_requires will NOT be automatically installed on the system where the setup script is being run. They are simply downloaded to the ./.eggs directory if they’re not locally available already. If you want them to be installed, as well as being available when the setup script is run, you should add them to install_requires and setup_requires.)
```

If that doesn't work the the dependency on Cython might need to be added to hidapi directly.

c1728p9, 2015-09-02:
I tried adding it to both install_requires and setup_requires and installation still failed.  I opened this ticket in hidapi to track this problem:
https://github.com/trezor/cython-hidapi/issues/17

bridadan, 2015-09-02:
Should be good to close now that this is merged: https://github.com/trezor/cython-hidapi/pull/18

closed 2015-09-02 by 0xc0170