pyocd/pyOCD #184: pyOCD style guide and tools

c1728p9, 2015-09-05:
To make contributions easier, pyOCD needs a recommended style guide/naming convention, along with tools to easily check these and auto formatting source.

My preference is that the code is compliant to pep8 where reasonable.  Contributors would be encouraged to follow the conventions, but would not be required to.  This could also be automated to post warnings on pull requests if the code has pep8 inconsistencies.

Development IDEs
-Eclipse with pyDev - allows auto formatting, and highlighting of code that is non-pep8 compliant.  It also is able to catch errors, such as referencing variables that doesn't exist
-pyCharm

Code style checking tools:
-pep8
-pylint

Auto formatting Tools
-autopep8

If anyone has other ideas or preferences please respond to this issue.

sg-, 2015-09-05:
+1 for pep8

0xc0170, 2015-09-05:
Agree, pep8 would be a good start to at least format the current code.

c1728p9, 2015-09-08:
What are your preferences on maximum line length?  pep8 prefers 79 characters, and if that is not feasible then 99 characters.  Limiting to 79 characters seems a little restrictive, but is definitely workable.  All my recent changes have been within the 79 character limit.  Either way, I don't think this should be a hard requirement, especially for things like tables that require wide spacing along with aligned columns for best readability.

0xc0170, 2015-09-09:
No rejecting a changeset just because it has a line with 90 characters :grin: Consistency and readability is important.