pyocd/pyOCD #188: python setup.py egg_info does very strange things

autopulated, 2015-09-23:
I get the error below when running a `pip install .` in an environment that tox has set up. I have absolutely no idea why `pgen` is being run when `python setup.py egg_info` is run in pyOCD, but I'm hoping someone here does?

```
Collecting pyyaml<4,>=3 (from valinor<1.0,>=0.0.0->yotta==0.7.1)
  Using cached PyYAML-3.11.tar.gz
Collecting pyOCD<1.0,>=0.3 (from valinor<1.0,>=0.0.0->yotta==0.7.1)
  Using cached pyOCD-0.5.1.zip
    Complete output from command python setup.py egg_info:
    usage: pgen [-h] [-v] [-q] [--version]
                {clean,build,init,import,export,list} ...
    pgen: error: invalid choice: '/var/folders/k_/yq4_rh2d05gbzxls2wr9r8mw0000gp/T/easy_install-amRspV/Cython-0.23.2/Cython/Parser/Grammar' (choose from 'clean', 'build', 'init', 'import', 'export', 'list')
    Traceback (most recent call last):
      File "<string>", line 20, in <module>
      File "/private/var/folders/k_/yq4_rh2d05gbzxls2wr9r8mw0000gp/T/pip-build-itqBec/pyOCD/setup.py", line 70, in <module>
        include_package_data=True,  # include files from MANIFEST.in
      File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/distutils/core.py", line 111, in setup
        _setup_distribution = dist = klass(attrs)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/dist.py", line 268, in __init__
        self.fetch_build_eggs(attrs['setup_requires'])
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/dist.py", line 313, in fetch_build_eggs
        replace_conflicting=True,
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/pkg_resources/__init__.py", line 836, in resolve
        dist = best[req.key] = env.best_match(req, ws, installer)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1081, in best_match
        return self.obtain(req, installer)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1093, in obtain
        return installer(requirement)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/dist.py", line 380, in fetch_build_egg
        return cmd.easy_install(req)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 629, in easy_install
        return self.install_item(spec, dist.location, tmpdir, deps)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 659, in install_item
        dists = self.install_eggs(spec, download, tmpdir)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 842, in install_eggs
        return self.build_and_install(setup_script, setup_base)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1070, in build_and_install
        self.run_setup(setup_script, setup_base, args)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/command/easy_install.py", line 1056, in run_setup
        run_setup(setup_script, args)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/sandbox.py", line 240, in run_setup
        raise
      File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/sandbox.py", line 193, in setup_context
        yield
      File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/contextlib.py", line 35, in __exit__
        self.gen.throw(type, value, traceback)
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/sandbox.py", line 164, in save_modules
        saved_exc.resume()
      File "/Volumes/Work/synced/Dev/autopulated/yotta/.tox/py27/lib/python2.7/site-packages/setuptools/sandbox.py", line 138, in resume
        type, exc = map(pickle.loads, self._saved)
      File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 1382, in loads
        return Unpickler(file).load()
      File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 858, in load
        dispatch[key](self)
      File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pickle.py", line 1133, in load_reduce
        value = func(*args)
    TypeError: __init__() takes at least 3 arguments (1 given)
    
    ----------------------------------------
Command "python setup.py egg_info" failed with error code 1 in /private/var/folders/k_/yq4_rh2d05gbzxls2wr9r8mw0000gp/T/pip-build-itqBec/pyOCD
```

0xc0170, 2015-09-23:
Is it trying to install cython? Uninstall project_generator and try again. 

I recall I found this there recently, created a tracking issue: https://github.com/project-generator/project_generator/issues/271, we might need to come-up with another name :(

autopulated, 2015-09-24:
yup if I change the installation order so that cython comes first things work :)

0xc0170, 2015-09-24:
I'll close this as it does not relate to pyOCD.

closed 2015-09-24 by 0xc0170