pyocd/pyOCD #189: Programming external boards with PyOCD / CMSIS

struempelix, 2015-09-24:
Hi there, 
I just switched the MCU firmware of my nrf51-DK from Jlink to mbed and figured out that I can't flash my external nrf51 based board anymore. This is not working by either using mass-storage or the pyocd-flashtool. In both cases the DK's nrf1 not the external board.
Is this the expected behavior? 
Other then that I can connect pyOCD to the DK, flash a hex file, use the pyocd-tool and so on. Great! I never got openOCD working on my MAC.

c1728p9, 2015-09-24:
Hi @struempelix 
Yeah, this is expected. The reason you can't flash external boards is because existing CMSIS-DAP software only uses the SWD pins connected to the nrf51.  There is another set of pins on Debug Out.  To debug an external board the CMSIS-DAP would need to detect that an external device has been attached and then switch SWD to use Debug Out.

struempelix, 2015-09-24:
Ok, good to know. I'll issue a feature request in the mbed CMSIS-DAP project. 
Thanks!!

c1728p9, 2015-10-08:
Closing this issue since it is unrelated to pyOCD

closed 2015-10-08 by c1728p9