pyocd/pyOCD #190: pyOCD/tools support for non-mbed boards

tgarc, 2015-09-27:
So I noticed that the `MbedBoard` class is hardwired into the tools so that there is no way to run the tools using a non-mbed board without changing the core code. This seems kind of silly since you can basically change two variables - the `mbed_vid` and `mbed_pid` - and everything else works as expected. Would it be possible to add something so that the user can specify vid/pid instead of hardwiring it?

Follow up question: Is the`MbedBoard` class really necessary? Or rather is it possible to merge that code into the `Board` class? It just seems like the *only* thing in that class that is really mbed specific in the class is the vid and pid. I think even the BoardInfo table could be included in the `Board` class without issue.

Perhaps the `MbedBoard` class could be dropped and the pid and vid could be arguments to the `Board` class with default values equal to the mbed device value which could be changed in the tools using command line arguments.

BTW, I'd be happy to implement this, but I'm just trying to get an idea if y'all would be supportive of taking the code in this direction before I actually code anything.

c1728p9, 2015-09-29:
Hi @tgarc 
Thanks for the interest in the project.  There are some changes planned in just this area.  The goal is to make pyOCD not tied to the VID/PID, but to still detect the attached CMSIS-DAP enabled debuggers.  If the board was an mbed then the target info could be easily extracted, otherwise the target would need to be detected or specified by the user.

Also, out of curiosity, what use case is giving you trouble?  Are you working on a custom CMSIS-DAP device with its own VID/PID?

tgarc, 2015-09-29:
@c1728p9 Thanks for the response. I'm glad to hear this is already in the works! You're right, I am using a CoLinkEx with the CMSIS-DAP firmware for debugging. I do have to provide the target to the command line tools, but that makes sense because I'm not using a pyOCD "supported" board. Works just fine other than that PID/VID restriction.

Is there anything planned along the lines of my questions about the `MbedBoard` class? The `getAllConnectedBoards`, `listConnectedBoards` and `chooseBoard` methods in particular seem like they could easily be moved to the `Board` superclass and would be generally useful to anyone wanting to write their own `Board` subclass.

c1728p9, 2015-09-29:
The future of the MbedBoard class is a little uncertain at this point.  The logic might very well end up in the Board class.  The code which lists connected boards will definitely be changed, with most of those functions being moved down into the transport layer.

tgarc, 2015-10-02:
I see. Then I guess it will be just to keep my personal projects forked for now. Feel free to close the issue.

closed 2015-10-02 by tgarc

c1728p9, 2016-03-15:
Just a heads up, the PR #229 removes the use of VID/PID.  Boards supporting CMSIS-DAP are now found based on the product string.