pyocd/pyOCD #192: BUG: `chooseInterface` is not a valid function

tgarc, 2015-10-02:
I noticed in the `Board` class that there is some interface initialization code that refers to a function called `chooseInterface` that is currently not part of the code base. Here is the relevant piece:

```python
class Board(object):
    """
    This class associates a target, a flash, a transport and an interface
    to create a board
    """
    def __init__(self, target, flash, interface, transport="cmsis_dap", frequency=1000000):
        if isinstance(interface, str) == False:
            self.interface = interface
        else:
            self.interface = INTERFACE[interface].chooseInterface(INTERFACE[interface])
```

I'm guessing this function just got taken out of the code at some point and was replaced with something else (`chooseBoard`?).

c1728p9, 2015-10-03:
Thanks for reporting.  Looking at the blame for that change, it looks like it was there from the beginning of the project and was a bug when it was checked in.  Feel free to open a PR to fix this if you have time.  Otherwise I'll remove it some time in the future.

0xc0170, 2015-10-05:
PR welcome!

tgarc, 2015-10-06:
OK, will submit that soon

PR #193 (2015-10-07): Removed reference to non-existent "chooseInterface" 
It doesn't seem easily possible to auto-initialize an `Interface` (interface initialization parameters are currently set inside the `getAllConnectedInterface` function inside the usb backend) so I decided to simply remove the auto-initialization code for the `interface` inside `Board.__init__`.

Closes #192
closed 2015-10-07 by 0xc0170