pyocd/pyOCD #202: semihost-telnet crashes when launching GDB server

bremoran, 2015-10-22:
pyOCD==0.6.0
Mac OS X 10.9.5
Python 2.7.9
arm-none-eabi-gdb 7.6.0.20131129-cvs

When I launch a debug session from yotta using yt debug, I get the following traceback immediately once the gdb session is connected:
```
Exception in thread semihost-telnet:
Traceback (most recent call last):
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/site-packages/pyOCD/target/semihost.py", line 327, in _server
    self.connected = self._abstract_socket.connect()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdb_socket.py", line 36, in connect
    self.init()
  File "/usr/local/lib/python2.7/site-packages/pyOCD/gdbserver/gdb_socket.py", line 31, in init
    self.s.bind(('', self.port))
  File "/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/lib/python2.7/socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 48] Address already in use
```

0xc0170, 2015-10-22:
I reference here the issue from valinor, as threading change was there earlier, might be a bug there: https://github.com/ARMmbed/valinor/issues/16

autopulated, 2015-10-22:
Note that the debug session should still work OK with this error, or you can revert to `pip install valinor==0.0.10` for now.

bremoran, 2015-10-22:
Yes, the debug session continues working.

flit, 2015-10-31:
Can this be closed now?

Btw, what was causing the socket to be reinited? Is there a bug in the `TelnetSemihostIOHandler` class?

0xc0170, 2015-11-02:
Resolved via #205 

Chris, the root cause has not been found yet

closed 2015-11-02 by 0xc0170