pyocd/pyOCD #203: Using websockets is not clear in pyocd-gdbserver

0xc0170, 2015-10-22:
I noticed today looking at the websocket support, that gdbserver provides port (-p) argument which as help says it's for port. But looking at the code, it is not just a port, it can be url address for websockets, code reference: https://github.com/mbedmicro/pyOCD/blob/master/pyOCD/gdbserver/gdbserver.py#L220

flit, 2015-10-22:
Yes, I've wondered about this, too. Was it intended to support debugging via the online IDE? I carried over the websocket support to semihosting just in case.

0xc0170, 2015-10-23:
> Was it intended to support debugging via the online IDE?

I would guess yes.

 we are currently playing with this websockets, we might share more soon

jupe, 2017-04-28:
This should be documented, and if there is some experimental features it is ok as long as its documented so.

flit, 2018-10-19:
Websocket support will be removed in a future version.

flit, 2019-03-04:
Closing since websocket support was removed in #551.

closed 2019-03-04 by flit