pyocd/pyOCD #208: Python 3 compatible

0xc0170, 2015-11-09:
We should make pyOCD compatible with python 3.

ntoll, 2016-12-12:
Yes, absolutely this.

Machine-Hum, 2018-07-03:
Is there an update on this?

flit, 2018-07-07:
@Machine-Hum @ntoll We're starting to work on it! The goal is to have Python 3 support by the end of this year. (There are a lot of other great features coming, too.)

flit, 2018-07-15:
Ok, the PR to add Python 3 support is now ready for testing: #374! Please try it out and report any issues you run into. All test cases pass, but I'd like to get some direct feedback from users.

Machine-Hum, 2018-07-16:
WOOT!

flit, 2018-07-18:
[Version 0.11.0](https://github.com/mbedmicro/pyOCD/releases/tag/v0.11.0) of pyOCD is now released with the Python 3 support, so you can update via pip.

Since Py3 support is official now, I'm closing this issue. If you do run into any problems, please create new issues. Thanks!

closed 2018-07-18 by flit