pyocd/pyOCD #210: gdbserver doesn't implement mon `init`

tgarc, 2015-11-13:
I noticed that my init sequence wasn't being run when I issued the ``mon init`` command from gdb. Looking into it further I found that the ``handleRemoteCommand`` of gdbserver doesn't implement ``init`` ( or ``reg``). This seems like a pretty bad bug, how was this not caught sooner? Am I missing something?

0xc0170, 2015-11-13:
These features are not yet supported.  I tagged it as a bug, as it could fail with a message as not supported.
Feel free to add it to gdbserver.

tgarc, 2015-11-13:
Ok, PR coming soon...

tgarc, 2015-11-17:
I've put something together, but I can't do a pull request because I can't get the latest committed source to run on my machine:

```
DEBUG:root:closing interface
DEBUG:root:init board <pyOCD.board.mbed_board.MbedBoard object at 0x7f3c740eb990>
DEBUG:root:cortexm init
DEBUG:root:closing interface
uncaught exception: 
Traceback (most recent call last):
  File "/user/tgarcia/git/pyOCD/pyOCD/tools/gdb_server.py", line 238, in run
    frequency=self.args.frequency)
  File "/user/tgarcia/git/pyOCD/pyOCD/board/mbed_board.py", line 238, in chooseBoard
    mbed.init()
  File "/user/tgarcia/git/pyOCD/pyOCD/board/board.py", line 54, in init
    self.target.init()
  File "wwolf_tools.py", line 126, in init
    CortexM.init(self)
  File "/user/tgarcia/git/pyOCD/pyOCD/target/cortex_m.py", line 363, in init
    self.link.connect()
  File "/user/tgarcia/git/pyOCD/pyOCD/pyDAPAccess/dap_access_usb.py", line 494, in connect
    assert isinstance(port, DAPAccessIntf.PORT)
AssertionError
```

Besides the fact that I had to install a new library to even get to that point. Can the latest commit be rolled back? It seems smart to keep the new pyDAPAccess stuff in a different branch or just not to merge the PR until it's been vetted.

c1728p9, 2015-11-17:
I did a lot of testing before I merged the latest commit so I'm surprised it is giving you trouble.  Testing GDB was part of these tests, and it didn't have any problems.  Do you have any local changes that could be interfering?  It looks like you might be using a modified version since wwolf_tools.py is in the callstack.

The line you are getting an assert on appears to be in the connect function.  The purpose of this is to check that either jtag or swd is specified as a port.  Did you change the default value of this call, or pass in a value that is not part of the PORT enum?

tgarc, 2015-11-17:
I will have to see what is going on when I get the next chance. My only modification was to go in an monkey patch the hardcoded mbed_vid/pid with the one for my debug adapter. I added some "hooks" for supporting a custom target, but it doesn't do any modifications to the core code. Anyway, I guess I'm a little frustrated that such a big change was added in one commit since I now have to debug that before I can add a patch that is unrelated. I hope that what I'm running into now is really minor.

tgarc, 2015-11-18:
Opened PR #211 to resolve this.

closed 2015-11-18 by tgarc

c1728p9, 2015-11-18:
Hi @tgarc did you find the cause of the assert?  If there is a problem with the pyDAPAccess refactoring I would like to address it sooner than later.

c1728p9, 2015-11-18:
@tgarc I tried your pull request locally, but was not able to reproduce your assert.  I did see a transfer error when running the init command through gdb.  This occurred because the read to 'IDCODE', which is required in the connect sequence, was not occurring a second time due to caching.  The code causing this is located here:
https://github.com/mbedmicro/pyOCD/blob/master/pyOCD/target/cortex_m.py#L501

Changing the function to this fixed the re-init problems:
```
    def readIDCode(self):
        """
        return the IDCODE of the core
        """
        self.idcode = self.dap.readDP(DP_REG['IDCODE'])
        return self.idcode
```

Let me know if this fixes your problem.  If it does, feel free to add it to your pull request.  Also, I don't mean to cause any frustration with my pull request.  I didn't see a good way to split it up into smaller commits that were still functional.  It I was able to within reason I would have :).

tgarc, 2015-11-20:
hey @c1728p9 just to follow up, I simply hadn't installed the pyDAPAccess package which was causing the assert error. (I didn't realize everything all that code had been moved to a new package already). It seems to work fine now. The only thing I would say is that I would've expected an import error which would've made it more clear that it was a dependency error.

0xc0170, 2015-11-20:
Great !

I don't follow, the import should have failed? We haven't replaced those imports pyOCD.pyDAPAccess  with pyDAPAccess module..

tgarc, 2015-11-20:
Wait, so are you saying I don't have to install pyDAPAccess separately? I'm not saying the import should've failed I'm just saying an import error is the kind of error I would expect when missing a dependency. But I've confused myself. Maybe installing pyDAPAccess from pip installed some dependencies that were indirectly causing this error.

c1728p9, 2015-11-23:
Unless you're using #209 there is no dependency on the pyDAPAccess package.  Also, having pyDAPAccess installed from pip shouldn't effect the current version of pyOCD.  Also, glad to hear my fix helped.