pyocd/pyOCD #212: pyOCD-gdbserver : [Errno 10013] An attempted access to a socket forbidden by its access permissions

adbridge, 2015-11-20:
Running the pyocd-gdbserver command in a gitbash shell running on Windows 7 Enterprise SP 1, 64 bit with a K64F attached via USB., the following error is observed. Running the latest pyOCD.

$ pyocd-gdbserver
INFO:root:new board id detected: 02400226a09a1e710000000000000000000000005d43e3c9
INFO:root:board allows 5 concurrent packets
INFO:root:DAP SWD MODE initialised
INFO:root:IDCODE: 0x2BA01477
INFO:root:K64F not in secure state
INFO:root:6 hardware breakpoints, 4 literal comparators
INFO:root:CPU core is Cortex-M4
INFO:root:FPU present
INFO:root:4 hardware watchpoints
INFO:root:Telnet: server started on port 4444
INFO:root:GDB server started at port:3333
ERROR:root:Unexpected exception: [Errno 10013] An attempt was made to access a socket in a way forbidden by its access permissions
Traceback (most recent call last):
  File "C:\Python27\lib\site-packages\pyOCD\gdbserver\gdbserver.py", line 313, in run
    connected = self.abstract_socket.connect()
  File "C:\Python27\lib\site-packages\pyOCD\gdbserver\gdb_socket.py", line 37, in connect
    self.init()
  File "C:\Python27\lib\site-packages\pyOCD\gdbserver\gdb_socket.py", line 32, in init
    self.s.bind((self.host, self.port))
  File "C:\Python27\lib\socket.py", line 224, in meth
    return getattr(self._sock,name)(*args)
error: [Errno 10013] An attempt was made to access a socket in a way forbidden by its access permissions

c1728p9, 2015-12-21:
Hi @adbridge , thanks for the report.  Is this error reproducible?  Also, do you have any other service running on port 3333 that could have prevented pyOCD from opening a socket?

0xc0170, 2015-12-21:
This is similar to what was reported around the same time, which was connected to valinor which introduced threading, and there was PR , which I'll have to find. I would assume this should be fixed on the master.

@adbridge Still valid?

c1728p9, 2015-12-22:
By chance I had this exact same problem happen to me today.  In my case it was because my VPN was using port 3333.  @adbridge , were you running a VPN when you saw this problem and if so does it go away when you use a different port?  You should be able to test this by running the following command 'pyocd-gdbserver -p 3334'.

adbridge, 2016-03-01:
@c1728p9  Sorry for the late reply, this got lost in my lnbox! No I wasn't running a VPN .
I just tried 'pyocd-gdbserver -p 3334' and got a different symptom:
$ pyocd-gdbserver -p 3334
INFO:root:new board id detected: 02400226a09a1e710000000000000000000000005d43e3c9
INFO:root:board allows 5 concurrent packets
INFO:root:DAP SWD MODE initialised
INFO:root:IDCODE: 0x2BA01477
INFO:root:K64F not in secure state
INFO:root:6 hardware breakpoints, 4 literal comparators
INFO:root:CPU core is Cortex-M4
INFO:root:FPU present
INFO:root:4 hardware watchpoints
INFO:root:Telnet: server started on port 4444
INFO:root:GDB server started at port:3334
The gitbash shell then just hangs and all I can do is <ctrl>C it to return to the command line

flit, 2016-03-01:
@adbridge That's the expected behaviour. The gdb server is waiting for a TCP connection from gdb. This implies that you do indeed have some other service using port 3333.

c1728p9, 2016-03-13:
Hi @adbridge , as flit mentioned, that is expected.  Are you still having issues using port 3333?

adbridge, 2016-03-15:
It looks like it seems to be working fine with port 3333 now. So not sure what the original problem was. Thanks.

c1728p9, 2016-03-15:
Closing since the issue has gone away.  Please re-open if you see it again.

closed 2016-03-15 by c1728p9