pyocd/pyOCD #217: Eclipse OpenOCD Debugging plug-in expects ``mon reg`` cmd

tgarc, 2015-12-09:
I noticed that the OpenOCD Debugging plug-in for Eclipse issues a ``mon reg`` on starting a debugging session. I'm guessing this is the reason for the no-op reg command that was removed by my last PR. This isn't high priority - I'm not sure yet how it effects debugging in Eclipse - I just wanted to document it.

Related to PR 211: https://github.com/mbedmicro/pyOCD/pull/211

c1728p9, 2015-12-21:
Thanks for opening an issue to investigate this further.  I have been using Eclipse with the OpenOCD plugin after #211 and haven't seen any problems.  @flit , do you know what the mon reg command is used for, if anything?

flit, 2015-12-22:
Afaik, the only purpose for the "reg" command is to dump register values. I'm pretty sure it doesn't hurt compatibility with the OpenOCD plugin if it is not implemented.

c1728p9, 2016-03-13:
Closing this issue since it isn't a problem.  If it becomes one please re-open.

closed 2016-03-13 by c1728p9