pyocd/pyOCD #222: Backtrace stopped: Cannot access memory at address XXXXXXXX

justbuchanan, 2016-02-23:
I'm trying to use pyOCD and gdb to debug some code running on my MBED LPC1768.  My code is segfaulting, but I'm not able to get anything useful out of gdb.  Here is the text output:

~~~{.sh}
Program received signal SIGSEGV, Segmentation fault.
0x0000c306 in HardFault_Handler ()
(gdb) bt
#0  0x0000c306 in HardFault_Handler ()
#1  <signal handler called>
#2  0x00019232 in osKernelStart ()
Backtrace stopped: Cannot access memory at address 0x1000801c
~~~

I'm launching pyocd and gdb with these commands:

~~~{.sh}
sudo pyocd-gdbserver --port 3333--reset-break --target lpc1768 -S -G

arm-none-eabi-gdb <path/to/elf/file> \
    -ex "target remote localhost:$(GDB_PORT)" \
    -ex "load" \
    -ex "continue"
~~~

Any suggestions for how I might get a more useful backtrace out of gdb?

adamgreen, 2016-02-23:
There are situations where GDB will get confused and give that error.  Sometimes it will try to walk too far up the stack when main() isn't found on the stack.  Stack corruption can cause it as well.

The only pyOCD thing you might want to try is adding a "monitor reset halt" command between the "load" and "continue" commands.

justbuchanan, 2016-02-27:
Thanks for the help!  I've done some investigating and I can confirm that the stack is getting corrupted, which explains why the backtrace isn't helpful.

closed 2016-02-27 by justbuchanan