pyocd/pyOCD #226: Unable to flash multiple devices simultaneously on Linux

VeliMattiLahtela, 2016-03-11:
Tested with newest Raspbian and Ubuntu 14.04 LTS.

Requirements: two open terminals and two K64F.
One device is flashing, trying to flash the other device produces this kind of error:

```
INFO:root:new board id detected: 0240000029164e45004d0012706e000ef301000097969900
Exception in thread Thread-2:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/interface/pyusb_backend.py", line 64, in rx_task
    self.rcv_data.append(self.ep_in.read(self.ep_in.wMaxPacketSize, -1))
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 975, in read
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 216, in setup_request
    self.managed_claim_interface(device, intf)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 16] Resource busy

Traceback (most recent call last):
  File "/usr/local/bin/pyocd-flashtool", line 9, in <module>
    load_entry_point('pyOCD==0.6.1', 'console_scripts', 'pyocd-flashtool')()
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/tools/flash_tool.py", line 150, in main
    frequency=args.frequency)
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/board/mbed_board.py", line 204, in chooseBoard
    all_mbeds = MbedBoard.getAllConnectedBoards(transport, False, blocking, target_override, frequency)
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/board/mbed_board.py", line 170, in getAllConnectedBoards
    mbed.write([0x80])
  File "/usr/local/lib/python2.7/dist-packages/pyOCD/interface/pyusb_backend.py", line 155, in write
    self.ep_out.write(data)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 387, in write
    return self.device.write(self, data, timeout)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 940, in write
    intf, ep = self._ctx.setup_request(self, endpoint)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 216, in setup_request
    self.managed_claim_interface(device, intf)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 102, in wrapper
    return f(self, *args, **kwargs)
  File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 167, in managed_claim_interface
    self.backend.claim_interface(self.handle, i)
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 811, in claim_interface
    _check(self.lib.libusb_claim_interface(dev_handle.handle, intf))
  File "/usr/local/lib/python2.7/dist-packages/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
usb.core.USBError: [Errno 16] Resource busy
```

This couldn't be reproduced on Windows.

c1728p9, 2016-03-14:
Thanks for the report @VeliMattiLahtela.  The pyDAPAccess PR has some fixes related to this.  Can you see if you still see this problem when using #209

VeliMattiLahtela, 2016-03-14:
The problem still persists.

c1728p9, 2016-12-06:
Hi @VeliMattiLahtela the PR #268 adds proper simultaneous device access support. Please test this PR and let me know if it fixes your problem.

EmbeddedAndroid, 2017-01-27:
@c1728p9 Hi Russ, I've just gotten around to testing this, and can confirm I'm able to have two devices concurrently now. Thanks!

c1728p9, 2017-01-30:
Closing this issue since is should be fixed now. Please reopen @VeliMattiLahtela if you see this issue again.

closed 2017-01-30 by c1728p9