pyocd/pyOCD #227: Continuous flashing of the same device freezes the flashing at some point on Linux

VeliMattiLahtela, 2016-03-11:
The following issue has been seen with newest raspbian and Ubuntu 14.04 LTS:

The binary file size is around 900Kb.

Running a bash script that repeats the same flashing command will at some point "freeze" the flashing.
The script I'm running:
```
#!/bin/bash
for i in `seq 1 10`;
do
   sudo pyocd-flashtool -d debug -b 02400226c36f3e5b0000000000000000000000003eb6c3e3 my_bin.bin
done   
```
I have not determined an exact amount of flashing times required before it freezes, seems to be quite random.

This is not reproducible on Windows.

I also noticed that adding "sudo pyocd-tool -b 02400226c36f3e5b0000000000000000000000003eb6c3e3 reset" before the flash command prevents the freeze.

Running the flash with debug doesn't provide any more information when the "freeze" happens.

I tried to backtrace the fail with no luck.

c1728p9, 2016-03-25:
That is very strange.  Are you running Ubuntu inside a virtual machine or natively?  Also, you are unable to get the python backtrace when this happens?

VeliMattiLahtela, 2016-04-08:
Ubuntu was run on VM.

c1728p9, 2016-04-09:
Was this inside virtual box?  It has some known USB driver bugs.  There was a PR to work around this a while back, but it never got merged:
https://github.com/mbedmicro/pyOCD/pull/154

VeliMattiLahtela, 2016-04-11:
Not with Virtual box, VMware Player was used.

c1728p9, 2017-02-11:
If it is the same issue as #154 this should be fixed by the timeout added [here](https://github.com/mbedmicro/pyOCD/blob/master/pyOCD/pyDAPAccess/interface/pyusb_backend.py#L139). If you are still seeing this hang please re-open with steps to reproduce.

closed 2017-02-11 by c1728p9