pyocd/pyOCD #230: Memory map - extension by core block regions

0xc0170, 2016-03-16:
If a user tries to read a memory from Private Peripheral Bus (PPB) (for cortex m4) (http://developer.arm.com/docs/dui0553/latest/4-cortex-m4-peripherals/41-about-the-cortex-m4-peripherals), it gets 

```
(gdb) x 0xE000E010
0xe000e010:     Cannot access memory at address 0xe000e010
```

It might be useful to extend memory map by additional core specific regions

Workaround: `` set mem inaccessible-by-default off ``

tgarc, 2016-03-16:
You're saying this happens when no memory region has been defined in gdb for the target? Doesn't this also happen with a generic cortex m3 device too?

adamgreen, 2016-03-16:
What about all of the memory regions for a device's peripherals? When writing my debug monitor I ended up with a number of different regions in my memory map to cover the PPB and the peripherals on the LPC1768: https://github.com/adamgreen/mri/blob/master/devices/lpc176x/lpc176x_init.c#L24

When adding new processors or using hardware debugging solutions like J-Link, I find ```set mem inaccessible-by-default off``` to be the simpler solution. I just add ```-ex "set mem inaccessible-by-default off"``` to the command line when I start arm-none-eabi-gdb

0xc0170, 2016-03-17:
@adamgreen Exactly what I was looking for to expand the memory maps to cover core peripherals, or just set that flag to off by default (our gdb server would set it off by default if not told otherwise). This is quite common to read outside of RAM and FLASH area for our targets

What do you guys think?

c1728p9, 2017-02-11:
Closing this since the GNU ARM Eclipse plugin already handles this from the user by setting
```
set mem inaccessible-by-default off
```

CC: @flit

closed 2017-02-11 by c1728p9