pyocd/pyOCD #236: pyOCD can not write FLASH of nRF52 devices

mjdietzx, 2016-05-13:
This is probably because it doesn't use or know about the non volatile memory controller and how to write flash on nRF52 devices.

Able to write RAM and read it back to verify.

When trying to write flash:
> target.writeMemory(0x100, 0x0)
> target.readMemory(0x0) # shows that value hasnt changed and is still erased.

c1728p9, 2016-05-13:
Hi @mjdietzx, writeMemory only works for writing to ram/peripherals.  Since ROM is read-only you can't write to it.

If you want to program images you can use the flash object.  Be aware though that it will erase the entire sector (since this is the smallest amount that can be erased).  The code will look something like this:
```
flash = board.flash
flash.flashBlock(addr, data)
```

Where address is a number and data is a bytearray or string.

mjdietzx, 2016-05-13:
OK, so flash.flashBlock() will erase any sectors it writes to by default? does this also apply to flash.flashBinary(..) if chip_erase is not set to True?

Is it within design goals of pyOCD to overwrite the writeu32() function in target_nrf52.py to correctly write to FLASH if addr is in FLASH memory map?

c1728p9, 2016-05-13:
Hi @mjdietzx, yeah, even if chip_erase is not set it will erase the sectors it writes to.

There aren't any plans to support flashing from the write functions.  Although, the nrf51/nrf52 treats flash like it is ram when flash write is enabled.  If you did this you would be able to write to it using the write function.  Writes will only complete correctly if you are writing to locations that have not already been written to though (you can only clear bits in flash without erasing a page).

Not sure if the addresses are the same for the nrf52, but at least on the nrf51 this should allow you to write to flash:
```
NVMC_CONFIG     = 0x4001E504
target.writeMemory(NVMC_CONFIG, 1)
```

closed 2016-05-14 by mjdietzx