pyocd/pyOCD #237: Cannot install from GitHub release zip

flit, 2016-05-15:
If you try to install pyOCD from the source zip downloadable from a [GitHub release](https://github.com/mbedmicro/pyOCD/releases/latest) (by running setup.py), you get the following error.

```
Traceback (most recent call last):
 File "setup.py", line 66, in <module>
   include_package_data=True,  # include files from MANIFEST.in
 File "C:\nxp\Python27\lib\distutils\core.py", line 111, in setup
   _setup_distribution = dist = klass(attrs)
 File "C:\nxp\Python27\lib\site-packages\setuptools\dist.py", line 272, in __in
it__
   _Distribution.__init__(self,attrs)
 File "C:\nxp\Python27\lib\distutils\dist.py", line 287, in __init__
   self.finalize_options()
 File "C:\nxp\Python27\lib\site-packages\setuptools\dist.py", line 327, in fina
lize_options
   ep.load()(self, ep.name, value)
 File "build\bdist.win-amd64\egg\setuptools_scm\integration.py", line 19, in ve
rsion_keyword
 File "build\bdist.win-amd64\egg\setuptools_scm\__init__.py", line 104, in get_
version
 File "build\bdist.win-amd64\egg\setuptools_scm\__init__.py", line 82, in _do_p
arse
LookupError: setuptools-scm was unable to detect version for 'C:\\Users\\B40219\
\Desktop\\FTF\\Mikroe\\Training\\pyOCD-0.7.0\\pyOCD-0.7.0'.

Make sure you're either building from a fully intact git repository or PyPI tarb
alls. Most other sources (such as GitHub's tarballs, a git checkout without the
.git folder) don't contain the necessary metadata and will not work.

For example, if you're using pip, instead of https://github.com/user/proj/archiv
e/master.zip use git+https://github.com/user/proj.git#egg=proj
```

Hoohaha, 2017-05-17:
@flit When are you going to release next version?

flit, 2018-10-02:
This was fixed in release v0.12.0.

closed 2018-10-02 by flit