pyocd/pyOCD #239: DAP TransferError when flashing board with pyOCD

kilogram, 2016-06-01:
When flashing an FRDM k64f with a wrapper script for pyOCD, I get the following exception: http://pastebin.com/nL4HBz0E

For a time after getting this error, I could not flash any binaries, even by mounting the device. I attempted to update the firmware to version 0226, only to run into the same problems. Running `pyocd gdbserver` produces a different but similar exception: http://pastebin.com/Pgqjwm7n

After updating the firmware to 0241 (https://github.com/mbedmicro/DAPLink/releases/tag/v0241), I am able to flash by mounting, but running the script or `pyocd gdbserver` produces the same error. Updating the version to the git master version, or running the most recently released version on pypi, produces the same error. By flashing the 0241 firmware and mounting the board as storage, I am able to flash the program, but not via pyOCD.

This happened to three k64f's. I'm not sure what triggered the issue; I was able to flash them at one point. I didn't update any relevant software before this started happening.

flit, 2016-06-01:
Which version of pyOCD are you using?

kilogram, 2016-06-01:
pyOCD==0.7.1
I also tried with the most recent master.

c1728p9, 2016-06-02:
Hi @kilogram, I ran into this issue as well.  It occurs when the K64F's flash is erased and left blank.  As a temporary workaround you should be able load another image over mass storage to recover the device.

flit, 2018-10-02:
Closing this issue. Problems related to blank flash on Kinetis devices should all be fixed now.

closed 2018-10-02 by flit