pyocd/pyOCD #241: pyocd connction error

MrH2S, 2016-06-07:
some error occurs after run 'yt debug'

There is the whole log info:
uncaught exception: The device has no langid
Traceback (most recent call last):
File "/usr/local/lib/python2.7/dist-packages/pyOCD/tools/gdb_server.py", line 250, in run
frequency=self.args.frequency)
File "/usr/local/lib/python2.7/dist-packages/pyOCD/board/mbed_board.py", line 190, in chooseBoard
target_override, frequency)
File "/usr/local/lib/python2.7/dist-packages/pyOCD/board/mbed_board.py", line 162, in getAllConnectedBoards
connected_daps = dap_class.get_connected_devices()
File "/usr/local/lib/python2.7/dist-packages/pyOCD/pyDAPAccess/dap_access_usb.py", line 347, in get_connected_devices
all_interfaces = getinterfaces()
File "/usr/local/lib/python2.7/dist-packages/pyOCD/pyDAPAccess/dap_access_usb.py", line 40, in getinterfaces
return INTERFACE[usb_backend].getAllConnectedInterface()
File "/usr/local/lib/python2.7/dist-packages/pyOCD/pyDAPAccess/interface/pyusb_backend.py", line 88, in getAllConnectedInterface
product = board.product
File "/usr/local/lib/python2.7/dist-packages/usb/core.py", line 841, in product
self.product = util.getstring(self, self.iProduct)
File "/usr/local/lib/python2.7/dist-packages/usb/util.py", line 314, in get_string
raise ValueError("The device has no langid")
ValueError: The device has no langid

0xc0170, 2016-06-07:
Please share the version of pyOCD you have, the firmware version of k64f interface ( I recall you had k64f board, correct?). state all the information that can help us to reproduce the problem

infinnovation, 2016-06-07:
I hope I'm not appearing to hijack this issue, but I have the same problem when running pyocd-gdbserver -l

The K64F I am using has firmware revision 0223 as per DETAILS.TXT

pyOCD 0.7.1
pyusb 1.0.0
Both libusb-1.0 and libusb-0.1 installed
Host platform is Scientific Linux 6 with python2.6
Part of "lsusb -v" output:
```
Bus 005 Device 004: ID 0d28:0204 NXP LPC1768
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               1.10
  bDeviceClass            2 Communications
  bDeviceSubClass         0 
  bDeviceProtocol         0 
  bMaxPacketSize0         8
  idVendor           0x0d28 NXP
  idProduct          0x0204 LPC1768
  bcdDevice            1.00
  iManufacturer           1 MBED
  iProduct                2 MBED CMSIS-DAP
  iSerial                 3 0240022352f86e52af2493ea
  bNumConfigurations      1
```

infinnovation, 2016-06-08:
I looked into this a bit further.  It appears that pyusb attempts to get the langids from each device but catches any USBError and stores an empty tuple instead.  In my case, the error was [Errno 13] Access denied (insufficient permissions).  Running pyocd-gdbserver as root allowed it to find the mbed board.

MrH2S, 2016-06-08:
@infinnovation 
Yes,thanks for your answer that there is really a access permission.
I can connect to the board after 'sudo'.

c1728p9, 2016-09-14:
It looks like #262 might have fixed this. Should this issue be closed?

c1728p9, 2017-02-11:
Closing this issue since there has been a PR to address it and since sudo is able to work around it. Please re-open if you are still having problems.

closed 2017-02-11 by c1728p9