pyocd/pyOCD #245: Improve initialisation speed when connecting to the board

arekzaluski, 2016-07-07:
Hello, after this Pull request : https://github.com/mbedmicro/pyOCD/pull/216 I can see that initialisation process while connecting to the board is longer than it used to be. The time difference is not that big when using usb interface. However it is noticeable longer when using new web socket interface. Is it possible to improve the speed? @flit @c1728p9

arekzaluski, 2016-07-11:
I wrote some tests to measure time differences. I was using Mac OS X 10.10.4 and FRDM-K64F board connected over USB. I measured a time that elapsed from starting gdb_server tool to start of GDBServer. Following times are in seconds. 

before:
0.191744
0.179748
0.178920
0.182037
0.194835
after: 
2.202233
2.201426
2.089389
2.088281
2.119333

It seems that more packets are being written/read during the connection with the board after adding multi cores support. It takes around 10 times longer. Is it possible to somehow decrease this time? The difference is even bigger when websocket interface is being used as a transport layer.

arekzaluski, 2016-07-27:
@flit @c1728p9 Any idea on how we can decrease this time?

c1728p9, 2016-07-27:
@arekzaluski it might be related to reading info from the ROM table. I would recommend profiling the code more to get an idea of which operations are taking a lot of time.

matthewelse, 2016-07-28:
@arekzaluski @c1728p9 @flit I've done some basic profiling, and narrowed the bottleneck down to the read_table method in the ROMTable class. As far as I can tell, the ROM table isn't currently being used: is it possible that this feature could be optional, or alternatively can we find faster methods for reading the table? (for example, making many reads at once, rather than repeatedly reading words one at a time)

arekzaluski, 2016-07-29:
@matthewelse Thank you for investigation. @flit @c1728p9 Can we do anything to fix it somehow? This delay is very big while using web sockets.

arekzaluski, 2016-08-08:
I've commented self.init_rom_table() line from ap.py and time is decreased to 0.499451 s on my Mac. @flit @c1728p9 Can we improve the time of reading rom table and/or make this feature optional?

c1728p9, 2016-09-14:
Initialization speed should be fixed with #249. Can this issue be closed @arekzaluski?

arekzaluski, 2016-09-15:
Fixed in #249. I'm closing this issue.

closed 2016-09-15 by arekzaluski