pyocd/pyOCD #246: pyocd-flashtool: ERROR:root:eraseAll error: 104 (Freescale KL25Z)

oliviermartin, 2016-07-21:
I am using pyOCD to flash the firmware onto the board. Since today, I cannot flash my KL25Z anymore. When I run the command `pyocd-flashtool --chip_erase -t kl25z` I have the error message: `ERROR:root:eraseAll error: 104`

```
pyocd-flashtool --chip_erase -t kl25z
INFO:root:DAP SWD MODE initialised
INFO:root:KL25Z not in secure state
INFO:root:2 hardware breakpoints, 0 literal comparators
INFO:root:CPU core is Cortex-M0+
INFO:root:2 hardware watchpoints
Erasing chip...
ERROR:root:eraseAll error: 104
Done
```

How to recover the board? What means the error code 104? I cannot find documentation about it.

flit, 2016-07-22:
@oliviermartin The error codes are from the Kinetis SDK 2.0 flash driver that underlies the flash algorithm. Error 104 is:
`````{c}
kStatus_FLASH_ProtectionViolation = MAKE_STATUS(
        kStatusGroupFlashDriver, 4), /*!< The program/erase operation is requested to execute on protected areas */
`````

This means you have protected one or more sectors using the Flash Config Field (FCF) located at address 0x400. You cannot perform a chip erase if you have protected sectors. If you need to write to the sectors that are protected, the only way you can recover is to perform a mass erase.

You can use pyocd-tool to perform the mass erase. After you connect with pyocd-tool, run the `unlock` command. This will perform a mass erase.

oliviermartin, 2016-07-26:
Thanks @flit, `pyocd-tool` actually solved my issue. I am not sure why the `unlock` parameter of `pyocd-flashtool` did not help me earlier.

closed 2016-07-26 by oliviermartin

flit, 2016-07-26:
@oliviermartin Well, the unlock param of pyocd-flashtool isn't implemented (yet).