pyocd/pyOCD #255: Large size of cmsis-svd 0.4

arekzaluski, 2016-08-12:
I was trying to install new pyocd that uses cmsis-svd 0.4 and I've noticed that 2 GB of disk space was not enough to install it. Should it really be that much bigger than version 0.3?

@posborne @flit @c1728p9

posborne, 2016-08-12:
2GB doesn't surprise me after decompression (the SVD XML compresses quite nicely actually) -- @flit added a bunch of new SVDs between 0.3 and 0.4 IIRC.  We've discussed splitting out the code from the SVD before but have avoided it thus far.

I could see a couple ways of splitting out the SVD:
1. Do not publish any SVDs with the python code and make people download it separately
2. Do not publish any SVDs with the python code but generate packages that can be installed for the various SVDs -- this could be decomposed by vendor, for instance.

I would welcome changes from you guys (I believe pyOCD is driving some of this) to achieve that.  You could get even more sophisticated and provide a way to grab SVDs dynamically at runtime as well (just grab them from Github ala https://github.com/rust-lang/crates.io-index).  Is there a rhyme/reason to the SVD files used by pyOCD?

flit, 2016-08-12:
Another option would be to store the SVDs in a zip, still included as part of cmsis-svd, which is accessed at runtime. This may be an easier short term solution.

For pyOCD, we just need the SVDs for supported target devices. This is currently a small subset of the total number of devices for which there are SVDs in cmsis-svd.

Ultimately I think the solution for pyOCD, at least, will be to use CMSIS-Packs. That may be the right way for cmsis-svd to dynamically load SVDs, too.

posborne, 2016-08-13:
> Ultimately I think the solution for pyOCD, at least, will be to use CMSIS-Packs

I'm on board with that.  When I starting pulling stuff together this was a documented thing but did not seem to be something that you could actually find out in the wild.  Maybe vendors have started to get their act together a little bit more.

arekzaluski, 2016-08-22:
Thanks for comments. Can this change be implemented? Without it it will be impossible to install pyOCD on systems with limited memory ( like online IDE ).

posborne, 2016-08-22:
If somebody wants to take a shot, I would be glad to review.  I probably do not have the time at the moment to prioritize making the change myself.  Once we have a plan, changes in place I am happy to publish new releases, etc.

c1728p9, 2016-09-13:
A release is scheduled for this week. With the current size I can't push a release to pypi without causing issues for a lot of users. I put up PR #264 as a temporary fix since it is likely a more complete fix will not make it in this week. @arekzaluski, can you let me know if this fixes your problem? If anyone has concerns with this PR please let me know.

CC: @flit, @posborne

arekzaluski, 2016-09-14:
@c1728p9 I've checked this fix and it resolves the issue for now. I'm now able to install pyOCD without any problems. Thanks.

c1728p9, 2016-09-14:
PR #264 has been merged. Should I leave it open to discuss potential long term fixes or close it?

arekzaluski, 2016-09-15:
@c1728p9 I think that current fix is more a workaround. I think that we should either leave this ticket open or create a new ticket for further conversation.

c1728p9, 2017-02-11:
Closing this since the problem of large install size has been addressed. Please create a new ticket for svd support if this is a feature that is needed.

closed 2017-02-11 by c1728p9