pyocd/pyOCD #259: pyocd binaries need sudo to work

aurabindo, 2016-08-17:
Hi,

Simply running pyocd-gdbserver on terminal gives me 

``` python
Traceback (most recent call last):    
  File "/usr/bin/pyocd-gdbserver", line 5, in <module>    
    from pkg_resources import load_entry_point    
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2927, in <module>    
    @_call_aside    
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2913, in _call_aside    
    f(*args, **kwargs)    
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2952, in _initialize_master_working_set    
    add_activation_listener(lambda dist: dist.activate())    
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 956, in subscribe    
    callback(dist)    
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2952, in <lambda>    
    add_activation_listener(lambda dist: dist.activate())    
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2513, in activate    
    for pkg in self._get_metadata('namespace_packages.txt'):    
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2503, in _get_metadata    
    for line in self.get_metadata_lines(name):    
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1477, in get_metadata_lines    
    return yield_lines(self.get_metadata(name))    
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1469, in get_metadata    
    return self._get(self._fn(self.egg_info, name))    
  File "/usr/lib/python2.7/site-packages/pkg_resources/__init__.py", line 1580, in _get    
    with open(path, 'rb') as stream:    
IOError: [Errno 13] Permission denied: '/usr/lib/python2.7/site-packages/protobuf-3.0.0b3-py2.7.egg/EGG-INFO/namespace_packages.tx
```
Same case with pyocd-tool, `sudo pyocd-tool --version` is `0.1.dev24`

flit, 2016-08-17:
You shouldn't need root permissions for any of the pyOCD tools on any system (OS X, Linux, Windows).

It looks to me like pkg_resources from setuptools is trying to init, but you have the protobuf package installed with root access only. This is not directly related to pyOCD.

aurabindo, 2016-08-17:
@flit thank you for the information, I fixed the protobuf issue, but now I have the issue as reported in #241 
I really need to use it without requiring root previleges, because its not possible to make eclipse run pyocd with root previleges easily.

c1728p9, 2016-09-14:
It looks like #262 might have fixed this. Should this issue be closed?

aurabindo, 2016-10-25:
Sure, please go ahead @c1728p9

c1728p9, 2016-10-25:
@aurabindo can you confirm that sudo is no longer needed?

closed 2017-02-11 by c1728p9