pyocd/pyOCD #270: pyocd-flashtool doesn't work at all with RPiv3

jupe, 2016-12-09:
`pyocd-flashtool` doesn't list any devices I've connected. Instead `mbedls` works fine. pyocd was clean clone from head of master and installed with command `python setup.py develop` .

This might be similar issue than #241, and there was mention that #262 should fix the problem, but I've still problem.

What I used:
```
~/pyOCD/ $ git rev-parse HEAD
48610ba219264d5306fa68bef2cd4ca89ef82067
$ uname -a
Linux raspberrypi 4.4.26-v7+ #915 SMP Thu Oct 20 17:08:44 BST 2016 armv7l GNU/Linux`
```

What happens:

```
~/pyOCD/ $ pyocd-flashtool -l -d debug
DEBUG:root:Exception getting product string: The device has no langid
DEBUG:root:Exception getting product string: The device has no langid
DEBUG:root:Exception getting product string: The device has no langid
DEBUG:root:Exception getting product string: The device has no langid
DEBUG:root:Exception getting product string: The device has no langid

$ mbedls
mbedls -j
[
    {
        "daplink_auto_reset": "0",
        "daplink_automation_allowed": "0",
        "daplink_daplink_mode": "Interface",
....
```

jupe, 2016-12-09:
same happens with `pyocd-gdbserver`:

```
$ pyocd-gdbserver -l
No available boards are connected
```

jupe, 2016-12-09:
verified that master works fine with windows:
```
C:\git\github\mbedmicro\pyOCD>pyocd-flashtool -l
0 => FRDM-K64F [k64f] boardId => 0240000029164e4500400012706e0005f301000097969900
```

jupe, 2016-12-09:
linux (virtualbox - ubuntu 16.04) seems to be also broken:

```
$ pyocd-flashtool -l -d debug
DEBUG:root:Exception getting product string: The device has no langid
DEBUG:root:Exception getting product string: The device has no langid
DEBUG:root:Exception getting product string: The device has no langid
No available boards are connected

$ mbedls -j
[
    {
        "daplink_auto_reset": "0", 
        "daplink_automation_allowed": "0", 
```

jupe, 2016-12-12:
running with `sudo` in linux seems to work. So issue here is just lack of documentation. Please add some mention that this needs `sudo` access when running in linux and introduction how to run without `sudo` .

0xc0170, 2016-12-13:
Referencing #267 (same similar problem, less information though, duplicated).

c1728p9, 2017-02-11:
Closing this issue since it is a duplicate of #267. Please re-open if this is not the case.

closed 2017-02-11 by c1728p9