pyocd/pyOCD #271: pyOCD developer set-up doesn't work with Python 2.7

ntoll, 2016-12-12:
Hi,

Followed the README verbatim. I had to install nose by hand (shouldn't this be in the `dev-requirements.txt`?). Ran the tests and get...

```
Ran 40 tests in 0.161s

FAILED (errors=25)
```

Either the test suite needs fixing or setup instructions are borked. :-(

N.

flit, 2016-12-14:
pytest is in `dev-requirements.txt`. It is required instead of nose. We just need to update the readme to use py.test instead of nosetests.

Hsiangkai, 2017-08-11:
I have updated README.rst in https://github.com/mbedmicro/pyOCD/pull/330.
Please help me to review it. I tried these test commands in my local machine.

flit, 2018-10-12:
Closing. Developer setup has been updated in the docs/DEVELOPERS_GUIDE.md.

closed 2018-10-12 by flit