pyocd/pyOCD #280: Target Support EFM32

pfried, 2017-03-09:
What are the steps to support a new platform? We do have several boards here and some are supported by pcOCD, some are not.

E.g. our EFM32 boards by Silicon labs are not supported.

How much effort is it to add support for this platform? How could one provide the support?

c1728p9, 2017-03-24:
Hi @pfried, do the EFM32 devices you mention have support in a cmsis-pack? If so, support for this target may come in with #284.

c1728p9, 2018-11-30:
@flit can you provide any information on when pack targets will be supported?

flit, 2018-12-03:
Hi @pfried, we typically rely on silicon partners and community members to add support for new targets. For broader target support, we are going to be moving to a CMSIS-Pack based configuration system (in addition to the existing built-in targets). The issue tracking this is #357. The goal is to have this completed by March 2019 (work has started).

pfried, 2018-12-12:
I think we will wait for a pack support. For our current development we "solved" this issue by using the Segger J-Link Server for gdb debugging.

flit, 2019-02-07:
Fyi, version 0.16.0 added pack support. You can download a pack from http://www.keil.com/dd2/pack/ and use the `--pack` command line argument to make the devices it defines available to pyOCD.

pelrun, 2019-03-12:
Can confirm this works with the EFM32G pack downloaded from https://developer.arm.com/embedded/cmsis/cmsis-packs.

flit, 2019-03-13:
Excellent, thanks for reporting your success!

flit, 2019-04-16:
Closing this issue since EFM32 targets are now supported via CMSIS-Packs.

With version 0.19.0, you can now add EFM32 support just by running `pyocd pack -i efm32`. That will install quite a few packs, so you probably want to be more specific about the part number.

closed 2019-04-16 by flit

pfried, 2019-10-15:
@flit about the installation of the packs: Do I then have to specify the full pack path for the commands or is there a name shortcut for the argument or are the recognized globally and if I specify a target it matches with the pack?