pyocd/pyOCD #291: frdmk82f gdb reported an warning when connecting

Hoohaha, 2017-04-11:
Start the pyocd-gdbserver at default port, then connect board by using gdb command: target remote localhost:3333, but there are several warnings:

`(gdb) target remote localhost:3333
Remote debugging using localhost:3333
warning: Architecture rejected target-supplied description
warning: while parsing target memory map (at line 3): Can't convert start="0x800
00000L" to an integer
0x00000000 in ?? ()
`
But the gdb-server show normal:
`INFO:root:DAP SWD MODE initialised
INFO:root:K82F25615 not in secure state
INFO:root:ROM table #0 @ 0xe00ff000 cidr=b105100d pidr=4000bb4c4
INFO:root:[0]<e000e000:SCS-M4 cidr=b105e00d, pidr=4000bb00c, class=14>
WARNING:root:Invalid coresight component, cidr=0x0
INFO:root:[1]<e0001000: cidr=0, pidr=0, component invalid>
INFO:root:[2]<e0002000:FPB cidr=b105e00d, pidr=4002bb003, class=14>
WARNING:root:Invalid coresight component, cidr=0xb1b1b1b1
INFO:root:[3]<e0000000: cidr=b1b1b1b1, pidr=b1b1b1b1b1b1b1b1, component invalid>

WARNING:root:Invalid coresight component, cidr=0x0
INFO:root:[4]<e0040000: cidr=0, pidr=0, component invalid>
INFO:root:[5]<e0041000:ETM-M4 cidr=b105900d, pidr=4000bb925, class=9, devtype=13
, devid=0>
INFO:root:CPU core is Cortex-M4
INFO:root:FPU present
INFO:root:6 hardware breakpoints, 4 literal comparators
INFO:root:4 hardware watchpoints
INFO:root:Telnet: server started on port 4444
INFO:root:GDB server started at port:3333
INFO:root:One client connected!
`

Then i input following command:
`(gdb)load "C:/hello_world.out"`
I checked the terminal of gdb-server, no progress information, and download is failure.
However download no error when using iar with cmsis-daplink.
Why it is failed with pyocd?
What does the warning mean?

0xc0170, 2017-04-11:
cc @flit

c1728p9, 2017-04-11:
Hi @Hoohaha, if you aren't already, you need to use the ARM version of gdb arm-none-eabi-gdb. This can be found [here](https://launchpad.net/gcc-arm-embedded).

flit, 2017-04-11:
The "WARNING:root:Invalid coresight component" lines reported by pyocd-gdbserver can be ignored. Those are caused by attempting to read the ITM, DWT, and TPIU ROM table entries when DEMCR.TRCENA is not asserted. This does not affect functionality, and is already fixed on master.

I checked the memory map in `target_MK82F256xxx15.py` and it seems ok.

Hoohaha, 2017-04-12:
Hi @c1728p9 , still not work.

Hi @flit, the warning is reported by gdb not server, the message as below:
 `warning: while parsing target memory map (at line 3): Can't convert start="0x800
00000L" to an integer
0x00000b00 in ?? ()`

flit, 2017-04-13:
What is the output from `arm-none-eabi-gdb -version` on your system?

Hoohaha, 2017-04-14:
`GNU gdb (GNU Tools for ARM Embedded Processors) 7.10.1.20160923-cvs
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word".`

flit, 2017-04-14:
Thanks, that's definitely a valid version, from the GNU ARM Embedded 5.4-2016q3 release. Same as what I'm using.

I'll see if I can find my FRDM-K82F board and try to reproduce your problem.

What version of pyOCD are you using? (Run `pyocd-gdbserver --version`)

Hoohaha, 2017-04-16:
0.8.1a1

Hsiangkai, 2017-05-23:
gdb could not recognize 'L' in hexadecimal number reported by pyOCD. I think it should strip 'L' from hex() value to ensure there is no such character in hexadecimal numbers.

To modify getXML() in pyOCD/core/memory_map.py should fix the bug.