pyocd/pyOCD #294: how to load with gdb server with FDMKL28 with DAPLINK

hakehuang, 2017-04-12:
Hi Sirs,

I try to use pyocd to download the FRDMKL28Z board. below is the command line

`python gdb_server.py -b 02900203C3403E730000000000000000000000003EBCC31B -p 3246` 

and it starts fine with below log

> Welcome to the PyOCD GDB Server Beta Version
INFO:root:DAP SWD MODE initialised
INFO:root:KW40Z4 not in secure state
INFO:root:ROM table #0 @ 0xf0002000 cidr=b105100d pidr=8e000
INFO:root:[0]<f0000000:MTB-M0+ cidr=b105900d, pidr=4001bb932, class=9, devtype=3
1, devid=0>
INFO:root:[1]<f0001000:MTBDWT cidr=b105900d, pidr=8e000, class=9, devtype=4, dev
id=0>
INFO:root:[2]<f0006000:CTI cidr=b105900d, pidr=4000bb9a6, class=9, devtype=14, d
evid=1040800>
INFO:root:[3]<e00ff000:ROM cidr=b105100d, pidr=4000bb4c0, class=1>
INFO:root:ROM table #1 @ 0xe00ff000 cidr=b105100d pidr=4000bb4c0
INFO:root:[0]<e000e000:SCS-M0+ cidr=b105e00d, pidr=4000bb008, class=14>
INFO:root:[1]<e0001000:DWT-M0+ cidr=b105e00d, pidr=4000bb00a, class=14>
INFO:root:[2]<e0002000:BPU cidr=b105e00d, pidr=4000bb00b, class=14>
INFO:root:CPU core is Cortex-M0+
INFO:root:2 hardware breakpoints, 0 literal comparators
INFO:root:2 hardware watchpoints
INFO:root:Telnet: server started on port 4444
INFO:root:GDB server started at port:3246

but when I try to load it with gdb

> $ arm-none-eabi-gdb.exe qca_demo.out
GNU gdb (GNU Tools for ARM Embedded Processors) 7.8.0.20150604-cvs
Copyright (C) 2014 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from qca_demo.out...done.
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x1c000140 in ?? ()
(gdb) load
Loading section A0 rw, size 0x140 lma 0x0
Loading section P3 ro, size 0x10 lma 0x400
Loading section P4 ro, size 0x17da0 lma 0x410
Error finishing flash operation

flit, 2017-04-12:
Hi @hakehuang, the log shows that pyOCD thinks this is a FRDM-KW40Z and not a FRDM-KL28Z. The version of DAPLink on your KL28 board is old (it is reporting ID 0290); we had to change its board ID to 0205 because 0290 was already taken by the FRDM-KW40Z.

Are you using a pre-production FRDM-KL28Z? I thought this was fixed for the final production run.

If you don't want to update the DAPLink firmware, you can use the `-t frdmkl28z` command line option to force the target to the correct board.

hakehuang, 2017-04-14:
@flit , you are right, I update to the latest daplink then it works now. Thanks

closed 2017-04-14 by hakehuang