pyocd/pyOCD #295: pyocd-gdbserver crashes on Raspberry Pi

jupe, 2017-04-21:
pyocd-gdbserver doesn't work at all in Raspberry Pi v3

ref: IOTSYST-2177

tried rasbian image:
```
pi@flashdev2:~ $ sudo pyocd-gdbserver --version
0.8.1a2.dev22
pi@flashdev2:~ $ sudo pyocd-gdbserver -d debug -p 3333 -b 0240000031754e45003c00189486002ade61000097969900
DEBUG:root:init board <pyOCD.board.mbed_board.MbedBoard object at 0x762070f0>
DEBUG:root:Started loading SVD
DEBUG:root:Completed loading SVD
INFO:root:DAP SWD MODE initialized
DEBUG:root:closing interface
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 810, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 763, in run
    self.__target(*self.__args, **self.__kwargs)
  File "build/bdist.linux-armv7l/egg/pyOCD/pyDAPAccess/interface/pyusb_backend.py", line 140, in rx_task
    self.rcv_data.append(self.ep_in.read(self.ep_in.wMaxPacketSize, 10 * 1000))
  File "build/bdist.linux-armv7l/egg/usb/core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "build/bdist.linux-armv7l/egg/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "build/bdist.linux-armv7l/egg/usb/backend/libusb1.py", line 851, in intr_read
    timeout)
  File "build/bdist.linux-armv7l/egg/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "build/bdist.linux-armv7l/egg/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 32] Pipe error

uncaught exception: Device 0240000031754e45003c00189486002ade61000097969900 read thread exited
Traceback (most recent call last):
  File "build/bdist.linux-armv7l/egg/pyOCD/tools/gdb_server.py", line 265, in run
    frequency=self.args.frequency)
  File "build/bdist.linux-armv7l/egg/pyOCD/board/mbed_board.py", line 265, in chooseBoard
    mbed.init()
  File "build/bdist.linux-armv7l/egg/pyOCD/board/board.py", line 50, in init
    self.target.init()
  File "build/bdist.linux-armv7l/egg/pyOCD/target/family/target_kinetis.py", line 62, in init
    self.mdm_ap.init(False)
  File "build/bdist.linux-armv7l/egg/pyOCD/coresight/ap.py", line 86, in init
    self.rom_addr = self.read_reg(AP_ROM_TABLE_ADDR_REG)
  File "build/bdist.linux-armv7l/egg/pyOCD/coresight/ap.py", line 100, in read_reg
    return self.dp.readAP((self.ap_num << APSEL_SHIFT) | addr, now)
  File "build/bdist.linux-armv7l/egg/pyOCD/coresight/dap.py", line 293, in readAP
    return readAPCb()
  File "build/bdist.linux-armv7l/egg/pyOCD/coresight/dap.py", line 284, in readAPCb
    result = result_cb()
  File "build/bdist.linux-armv7l/egg/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 652, in read_reg_cb
    res = transfer.get_result()
  File "build/bdist.linux-armv7l/egg/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 119, in get_result
    self.daplink.flush()
  File "build/bdist.linux-armv7l/egg/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 583, in flush
    self._read_packet()
  File "build/bdist.linux-armv7l/egg/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 736, in _read_packet
    raw_data = self._interface.read()
  File "build/bdist.linux-armv7l/egg/pyOCD/pyDAPAccess/interface/pyusb_backend.py", line 203, in read
    self.serial_number)
DeviceError: Device 0240000031754e45003c00189486002ade61000097969900 read thread exited
```

and with archlinux:
```
[alarm@alarm ~]$ sudo pyocd-gdbserver --allow-remote
INFO:root:DAP SWD MODE initialized
INFO:root:K64F not in secure state
[... 66 lines cut ...]
  File "build/bdist.linux-aarch64/egg/pyOCD/pyDAPAccess/dap_access_cmsis_dap.py", line 377, in _decode_transfer_block_data
    raise ValueError('DAP_TRANSFER_BLOCK response error')
ValueError: DAP_TRANSFER_BLOCK response error
INFO:root:Client detached
INFO:root:Telnet: server stopped
Exception in thread Thread-1:
Traceback (most recent call last):
  File "/usr/lib/python2.7/threading.py", line 801, in __bootstrap_inner
    self.run()
  File "/usr/lib/python2.7/threading.py", line 754, in run
    self.__target(*self.__args, **self.__kwargs)
  File "build/bdist.linux-aarch64/egg/pyOCD/pyDAPAccess/interface/pyusb_backend.py", line 140, in rx_task
    self.rcv_data.append(self.ep_in.read(self.ep_in.wMaxPacketSize, 10 * 1000))
  File "build/bdist.linux-aarch64/egg/usb/core.py", line 402, in read
    return self.device.read(self, size_or_buffer, timeout)
  File "build/bdist.linux-aarch64/egg/usb/core.py", line 988, in read
    self.__get_timeout(timeout))
  File "build/bdist.linux-aarch64/egg/usb/backend/libusb1.py", line 851, in intr_read
    timeout)
  File "build/bdist.linux-aarch64/egg/usb/backend/libusb1.py", line 936, in __read
    _check(retval)
  File "build/bdist.linux-aarch64/egg/usb/backend/libusb1.py", line 595, in _check
    raise USBError(_strerror(ret), ret, _libusb_errno[ret])
USBError: [Errno 110] Operation timed out
```

jupe, 2017-04-28:
This might be actually RPiv3 usb driver issue and nothing to do with pyOCD, but that is current feeling based on tests that we did. Anyway, I propose to mention in readme/documentation that pyOCD doesn't work with RPi.

flit, 2018-10-12:
Hi @jupe, is still issue still a problem with the latest pyOCD and RasPi software?

jupe, 2018-11-23:
pyocd@v0.12.0 pyocd-gdbserver does not start at all for some reason - or it takes more than 20 seconds to print anything. 

pyocd@v0.13.2 is more promising:
```
$ pyocd-gdbserver --version
0.13.2
$ pyocd-gdbserver -l
No available debug probes are connected
$ pyocd-gdbserver 
Waiting for a debug probe to be connected...
```

But not sure why debug probe is not connected, any idea ? mbedls show devices just fine:
```
$ mbedls
| platform_name | platform_name_unique | mount_point                                           | serial_port  | target_id                                        | daplink_version |
|---------------|----------------------|-------------------------------------------------------|--------------|--------------------------------------------------|-----------------|
| K64F          | K64F[0]              | /mnt/platform-3f980000_usb-usb-0_1_5_1_0-scsi-0_0_0_0 | /dev/ttyACM1 | 0240000033514e4500205015846c00256cb1000097969900 | 0243            |
| K64F          | K64F[1]              | /mnt/platform-3f980000_usb-usb-0_1_4_1_0-scsi-0_0_0_0 | /dev/ttyACM0 | 0240000034544e45002a00018aa9001d2011000097969900 | 0244            |
```

flit, 2018-11-24:
Try running it with debug logging enabled (`pyocd-gdbserver -l -ddebug`). It might be related to USB device permissions. I don't know if Raspbian or Archlinux use udev, but on other Linuxes with udev you need to add rules to enable user access to USB devices. See the [README.md](/mbedmicro/pyOCD/blob/master/README.md) for info about configuring udev.

No idea why 0.12.0 wouldn't start.

flit, 2019-02-02:
@jupe Have you had a chance to test with the latest release and with debug logging? The command to list probes with debug logging would now be `pyocd list -v`.

jupe, 2019-02-02:
RPi is not anymore needed for us.  Feel free to close ticket. Maybe its not worth of testing manually if there is no any plans to add rpi for CI...

flit, 2019-02-03:
Thanks for the update. I'm going to keep this issue open for now, as I would like to have pyOCD working on Raspberry Pi.

rmu75, 2019-07-30:
Just FYI, after some woes regarding installation of cmsis-pack-manager, and a separate issue with USB which I still have to investigate, pyocd seems to run fine on a Raspberry Pi, ubuntu 18.04.2.

closed 2026-01-07 by TeoMahnic