pyocd/pyOCD #299: starting gdb server with port 0 doesn't print actual port in use

jupe, 2017-04-28:
## Description
I want to start multiple gdb servers parallel. Now only way is to detect free port before calling `pyocd-gdbserver` because if I use `-p 0` it doesn't give me actual port number in use.

## How to reproduce
run gdb server
```
$ pyocd-gdbserver -p 0
...
INFO:root:GDB server started at port:0
```
pyocd is running now but it just print `..port:0` -which are not valid. See below:

```
$ lsof -iTCP -sTCP:LISTEN
COMMAND     PID     USER   FD   TYPE             DEVICE SIZE/OFF NODE NAME
Python    13091 jva    3u  IPv4 0x73fd86420b3627b3      0t0  TCP localhost:krb524 (LISTEN)
Python    13091 jva    4u  IPv4 0x73fd86420b36f7b3      0t0  TCP localhost:51418 (LISTEN)
```

jupe, 2018-11-20:
any chance to implement this. This blocks test beds integration... 
cc: @studavekar @kjbracey-arm

kjbracey-arm, 2018-11-20:
Would you care to test #455?

PR #455 (2018-11-26): Make "port 0" work properly as automatic
Specifying port zero on the command line resulted in an automatic port
selection, but didn't quite work well enough to be useful.

Refine behaviour so that:

* We print the port that was auto-selected in the debug.
* We re-use the same port when persisting to another session.
* We auto-select for every core, rather than trying to use ports
  1,2,3... for subsequent cores.

Fixes #299.
closed 2018-11-26 by flit