pyocd/pyOCD #301: disable telnet when starting pyocd-gdbserver

jupe, 2017-04-28:
## Description
When user doesn't need to use telnet server there should be way to disable it. It just waste of cpu&memory (& one extra socket) which might be problematic in "low-end-hosts" like `odroid`.

jupe, 2018-11-21:
@kjbracey-arm would it be hard to implement this as well? :)

kjbracey-arm, 2018-11-26:
This should be achievable by either turning off the `enable_semihosting` session option, or by setting `semihost_console_type` to `stdx`.

(Semihosting routes I/O calls on the device to the debugger machine - eg `printf` or file accesses. There are some hooks for this in Mbed OS, but it's not normally used, so disabling should normally be fine).

flit, 2019-03-07:
This can currently be achieved by setting `semihost_console_type` to something other than "telnet". 

Setting `enable_semihosting` to False does not work. The semihosting objects are always created. The enable flag just determines whether halts while running are examined for semihosting breakpoints. It's done this way to support the "arm semihosting enable" gdb monitor command to turn enable semihosting after the gdbserver is already running, mostly for OpenOCD compatibility.