pyocd/pyOCD #305: sudo needed to run on linux again

LuisMendoza94, 2017-05-11:
Hi, 
After the commit number 9ffdc69f07d4f3c984ad5434e4b18602a6117e59,
The issues #259, #241, #270, and #262 where re introduced.
This issue is caused by libusb, and therefore pyusb, requiring sudo to work.

A work around to this issue can be seen in this [stackoverflow](http://stackoverflow.com/a/32022908/6648143)

Did the commit 7515edd823166844d606110bcc6f2e30f132734a affected the following commits functionality that was necessary to remove the fix or is this just a merge conflict?

LuisMendoza94, 2017-05-11:
I reviewed the 9ffdc69f07d4f3c984ad5434e4b18602a6117e59 it does affect the previous solution, 
Is it possible to add this issue to the project  README

c1728p9, 2017-05-11:
Hi @LuisMendoza94, it sounds like this was a regression. I'll look into fixing it so a readme isn't necessary.

flit, 2018-10-12:
Closing this issue. The readme has been updated with information about udev rules, and an example udev rule file is included (`udev/50-pyocd.rules`).

closed 2018-10-12 by flit